sn4k3/UVtools #187: [FEATURE REQUEST] Mac ARM version

aaaldo, 2021-04-23:
## Is your feature request related to a problem? Please describe
This feature doesn't really address a problem. It's more of an optimisation : as the current Mac lineup is gradually transitioning to their in-house designed ARM chip, UVTools is still only built for x86 architectures, and thus requires an additional step of emulation to run on the newer Mac.

## Describe the solution you'd like
Make an ARM version of UVTools, much like the Linux version which is available for both x86 and ARM arch.

## Describe alternatives you've considered
Currently, the app is running fine using the x86 emulation layer

## Screenshots
Not needed

## Additional context
-

sn4k3, 2021-04-23:
Not ATM, we need to wait for .NET 6.0 release and hope that UI framework also support it

sn4k3, 2022-02-26:
Anyone here with M1 willing to test and build opencv on it for me?

aaaldo, 2022-02-26:
Hello @sn4k3 I do have a M1 and some spare time. I'll glad to help !

sn4k3, 2022-02-26:
Cool! 
Please test this package: https://we.tl/t-Bvm2zedmJI
It's UVtools compiled to arm64, but opencv is still x64, lets test this first
Then i will send you compile commands for opencv to generate the arm lib

aaaldo, 2022-02-26:
Hello @sn4k3 unfortunately, this doesn't launch at all. I've got a generic error, can't find anything in the logs as well.

sn4k3, 2022-02-26:
Check which ffmpeg you have installed: `ffmpeg --version`

aaaldo, 2022-02-26:
I'm afraid i'm getting the same error : https://gist.github.com/aaaldo/0ce1c5bdb0ec2f32135a914c869fb894

I did check my ffmpeg version, and it's now an older one : 

> ffmpeg version 4.4.1 Copyright (c) 2000-2021 the FFmpeg developers
> built with Apple clang version 13.0.0 (clang-1300.0.29.3)
> configuration: --prefix='/opt/homebrew/Cellar/ffmpeg@4/4.4.1' --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags= --enable-avresample --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libbluray --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox
> libavutil      56. 70.100 / 56. 70.100
> libavcodec     58.134.100 / 58.134.100
> libavformat    58. 76.100 / 58. 76.100
> libavdevice    58. 13.100 / 58. 13.100
> libavfilter     7.110.100 /  7.110.100
> libavresample   4.  0.  0 /  4.  0.  0
> libswscale      5.  9.100 /  5.  9.100
> libswresample   3.  9.100 /  3.  9.100
> libpostproc    55.  9.100 / 55.  9.100
>

sn4k3, 2022-02-26:
But in logs it still says: `/opt/homebrew/Cellar/ffmpeg/5.0/include/libavcodec/avcodec.h:2581:41: note: passing argument to parameter 'avctx' here`

Reboot your pc
Delete emgucv folder and repeat all steps from the git clone step

aaaldo, 2022-02-27:
Hello @sn4k3 compilation went a little further, but now it stopped again with this error :

>  /usr/local/share/dotnet/sdk/6.0.200/Sdks/NuGet.Build.Tasks.Pack/build/NuGet.Build.Tasks.Pack.targets(221,5): error NU5019: Fichier introuvable : '..\..\..\libs\Emgu.CV.Platform.Mac.dll'. [/Users/ibi/Downloads/temp/emgucv/Emgu.CV.Platform/NetStandard/Emgu.CV.Platform.NetStandard.csproj]
> make[2]: *** [platforms/nuget/Emgu.CV.runtime.macos/CMakeFiles/Emgu.CV.runtime.macos.nuget] Error 1
> make[1]: *** [platforms/nuget/Emgu.CV.runtime.macos/CMakeFiles/Emgu.CV.runtime.macos.nuget.dir/all] Error 2
> make: *** [all] Error 2

sn4k3, 2022-02-27:
Those files doesn't matter.
Check for `libcvextern.dylib` presence on `/emgucv/libs`

aaaldo, 2022-02-27:
Hello @sn4k3 I indeed have this file compiled.

I pasted it in the compiled app as suggested, yet i'm still having an unknown error without any logs.

sn4k3, 2022-02-27:
If you run UVtools from terminal, what does it says?

aaaldo, 2022-02-27:
I have this error using Terminal : 

> The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x600000deb630 {Error Domain=NSPOSIXErrorDomain Code=153 "Unknown error: 153" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}

aaaldo, 2022-02-27:
Here's the compiled lib : https://we.tl/t-O4hiapGhxA

sn4k3, 2022-02-27:
That's good news!

Is there a way to know if it is running on rosseta or not?

Also can you use my previous zip package and run via sh to see if x64 libcvextern runs in that mix?

aaaldo, 2022-02-27:
I've checked and it's indeed running natively 🎉 Congratulations ! 

I've also checked the previous zip package and it isn't working even using the .sh command. I have this error instead : 

> sh /Applications/UVtools.app/Contents/MacOS/UVtools.sh
> /Applications/UVtools.app/Contents/MacOS/UVtools.sh: line 3: 88364 Killed: 9               ./UVtools
> Unhandled exception. System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
>  ---> System.DllNotFoundException: Unable to load shared library 'cvextern' or one of its dependencies. In order to help diagnose loading problems, consider setting the DYLD_PRINT_LIBRARIES environment variable: dlopen(libcvextern, 0x0001): tried: 'libcvextern' (no such file), '/usr/local/lib/libcvextern' (no such file), '/usr/lib/libcvextern' (no such file), '/Applications/UVtools.app/Contents/MacOS/libcvextern' (no such file), '/usr/local/lib/libcvextern' (no such file), '/usr/lib/libcvextern' (no such file)
>    at Emgu.CV.CvInvoke.RedirectError(CvErrorCallback errorHandler, IntPtr userdata, IntPtr prevUserdata)
>    at Emgu.CV.CvInvoke..cctor()
>    --- End of inner exception stack trace ---
>    at Emgu.CV.CvInvoke.Init()
>    at UVtools.WPF.App.OnFrameworkInitializationCompleted() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\App.axaml.cs:line 169
>    at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
>    at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
>    at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run() in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 166
>    at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
>    at Avalonia.Native.PlatformThreadingInterface.SignaledCallback.Signaled(Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 39
>    at Avalonia.Native.Interop.Impl.__MicroComIAvnSignaledCallbackVTable.Signaled(IntPtr this, Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/Interop.Generated.cs:line 3788
> --- End of stack trace from previous location ---
>    at Avalonia.Native.PlatformThreadingInterface.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 90
>    at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
>    at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
>    at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
>    at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 59
> /Applications/UVtools.app/Contents/MacOS/UVtools.sh: line 3: 88365 Abort trap: 6           dotnet UVtools.dll

sn4k3, 2022-02-27:
Here the compiled package with your lib: https://we.tl/t-GTfNecci4H
Does it work without sh?

aaaldo, 2022-02-28:
It's still KO without sh.

I don't notice any speed improvements in the usual process i typically use. I did a test with a pretty big file (4hr long print, with almost a full 6" bed) and issues analyse and repair took almost exactly the same time.

However, memory consumption went drastically down from 1,33Go to just 554mo ! Also worth mentioning that the app feels more responsive overall, and launches almost instantly now.

sn4k3, 2022-02-28:
> It's still KO without sh.

I have to see if the package bundle is made same way as the x64, something isn't right.

> I don't notice any speed improvements in the usual process i typically use. I did a test with a pretty big file (4hr long print, with almost a full 6" bed) and issues analyse and repair took almost [...]

Ok thats good, overhead of emulation is always heavier.

sn4k3, 2022-02-28:
Can you zip an native light app and send me so i can see the contents? I'm not finding any information about arm64 bundles

sn4k3, 2022-02-28:
Test this one: https://we.tl/t-kXGs9lbaqB

aaaldo, 2022-02-28:
Nope, still the same error message :

> The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 

> "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x600001dd43c0 {Error Domain=NSPOSIXErrorDomain Code=153 "Unknown error: 153" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}

aaaldo, 2022-02-28:
> Can you zip an native light app and send me so i can see the contents? I'm not finding any information about arm64 bundles

Maybe you can take a look at Signal app : it's open source, and they have now universal builds : https://github.com/signalapp/Signal-Desktop

sn4k3, 2022-02-28:
I compared with this which is simpler and M1 based: https://github.com/macmade/Hot
Still i can't find anything wrong with my bundle.
Try to chmod UVtools:

`chmod +x UVtools.app/Contents/MacOS/UVtools`

Any change?

aaaldo, 2022-02-28:
Nope, still the same error :(

sn4k3, 2022-02-28:
Dam, i'm out of ideias :( 
Can't find any valid information on internet about that error

aaaldo, 2022-02-28:
I'm on macOS 12.2.1 (21D62)

> Dam, i'm out of ideias :( Can't find any valid information on internet about that error.
> 
> What OS do you run?

sn4k3, 2022-03-14:
osx-arm64 is now available

closed 2022-03-14 by sn4k3