sn4k3/UVtools #188: Calibration Exposure Creating Faulty Layer at interface [BUG]

dgm3333, 2021-04-24:
The calibration test is not creating consistent feature sizes at different layers as per the following screenshot.
This is slicing with 0.1mm layer height showing the layer at the interface (too thin) and the one above (OK).

The error isn't present when slicing 0.01 layer height (but on my print at that resolution many features are totally failing to adhere at that interface, so there's another issue I haven't figured out yet). Slower lift speed didn't help either (incidentally is it worth having a lift speed calibration option?

[image]

sn4k3, 2021-04-24:
It looks like you are using erodes? Can you post a screenshot of the values you use on the dialog?

dgm3333, 2021-04-24:
If I am using erodes I'm not aware of it, and the Erode Bottom setting is definitely 0. Is there anywhere else it would be set?
As noted the "erosion" doesn't occur with 1mm base thickness although all other settings were identical.
I'm unlikely to actually print the test any more since it isn't working for me.

[image]

sn4k3, 2021-04-24:
The Chamfer layers is also an erode, so you have 5 chamfered layers, it will cath layer 4 and 5 with features, set that to 3 or 0

closed 2021-04-30 by sn4k3