sn4k3/UVtools #193: [BUG]

gessima, 2021-05-06:
<span style="color:red">**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**</span>

## Describe the bug
A clear and concise description of what the bug is.
Calculator tool bottom light off delay does not work proper - set parameters on file is taking wrong parameters (from normal light off delay)
## To Reproduce
Steps to reproduce the behavior:
1. Go to 'Tools'
2. Click on 'Calculator'
3. Click on "light off delay"
3. Scroll down to 'Set parameters to file'
4. See error

## Expected behavior
A clear and concise description of what you expected to happen.
When setting 15sec to "bottom light off delay" and pressing button "set parameters to file" it takes the other parameter set in "light off delay"
## Screenshots
If applicable, add screenshots to help explain your problem. Never hide information, full GUI/window screenshot is always a must!

## Files
Were possible attach the problematic file, both source and output file if applicable.

## System
**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10 x64 v2004]
  
## Exceptions Log file
In some unexpected exceptions errors will generate and save a log file under appfolder/Logs/AppErrors.
If present you can attach the log file here. Check file name against date/time to be sure it's related to your problem.

## Additional context
Add any other context about the problem here.

sn4k3, 2021-05-06:
Works for me... What is your printer?

gessima, 2021-05-08:
I did not put the file to my printer - have just seen on the screen that the values are not taken correctly.  I load a file and in the bottom line there is the information Light off 1s/1s .  When I go to tools/calculator/Light off delay then I can choose to set the light off delay (left hand button) - or set the bottom light off delay (right hand button) . Now I set a value of 5s for the normal light off delay (left side) and a value of 10s for the bottom light off delay (right side) - and when I press now the left button to set the parameter to file (5s) I can see that the value in the file information that was before 1s/1s now is changing to 1s/5s . Seems to be be maybe correct.....When I press the right button to set the parameter to file (10s - for the bottom) the file information is changing to 5s/5s - and this is wrong !? Should be 10s/5s ??

gessima, 2021-05-08:
I have at work Anycubic Mono X and at home Anycubic Zero

sn4k3, 2021-05-08:
AnyCubic series don't have bottom* properties just the normal.
UVtools repeat the value from normal to show the bottom, and if you try to set the bottom it sets to the normal to keep compability with the program logic.

[image]

Note that's a firmware lacking of that settings and you can't have different settings for bottom beside the exposure time.
If you open the Edit print paramaters it will show you all the avaliable properties in the format to change.

Still i should disable bottom inputs at calculator if unable to, i will improve that

closed 2021-05-08 by sn4k3