sn4k3/UVtools #201: [BUG]

brettvitaz, 2021-05-15:
## System
 - OS: Mac OS 11.3.1 (Big Sur)
 - Slicer: Lychee
 - Printer: Elegoo Saturn

## Describe the bug
Calibration tools windows are not resizable and some content cannot be accessed. 

## To Reproduce
Steps to reproduce the behavior:
1. Click "Calibration" menu.
2. Select item "Elephant Foot".
3. Elephant foot window is fixed in size and items on the right are invisible and unaccessible.

## Expected behavior
Window should fit content, or allow scrolling.

## Screenshots
[image]

sn4k3, 2021-05-15:
BigSur was wrong information to framework. Go settings and disable this: 
[image]

brettvitaz, 2021-05-15:
Thank you for the quick response. I have also noticed that the underlying issue causes problems attempting to resolve the issue as seen in this screenshot:

[image]

The "Save Settings" button is not shown unless you navigate to one of the other settings tabs. Switching to the "Layer Preview" tab and back caused the Save button to appear.

sn4k3, 2021-05-15:
Yeah you can also hit ENTER.
Does it solve your problem?

brettvitaz, 2021-05-15:
Yes. The issue is resolved. Thank you

closed 2021-05-15 by brettvitaz

brettvitaz, 2021-05-15:
As I mentioned, my original issue is resolved, but I did some additional testing and noticed that hitting "Enter" toggles the state of the selected option, and does not save the settings. 

Note: Clicking on the tab title after setting the appropriate selection allows the user to then hit "Enter" to save, but I would not consider this an obvious course of action.

sn4k3, 2021-05-15:
Enter will not work on inputs where it can accept enter to do an action, eg: Open dropdown, OR in a text input with multiple lines where you need enter to do a line break. You need to click on window background so it gain focus and then ENTER will work.
This is not a bug with UVtools, but apple doing it's magic and reporting bad information about resolution

brettvitaz, 2021-05-15:
I completely understand this and have dealt with comparable limitations when developing user interfaces. I appreciate all of the work put into UVTools and I am only hoping to make it easier for other people who find the same issue to resolve.

While perhaps not a bug, it is a user experience problem that is present in UVTools software. If this project allows external contribution, I would be happy to look into ways to mitigate the issue.

sn4k3, 2021-05-15:
UVtools is open source, you can always fork it and modify.
The only way to fix this is auto disable that setting when detecting a macOS bigSur or greater