sn4k3/UVtools #202: [BUG] Windows Defender flags UVtools as Trojan:Win32/Emali.A!cl

MattCarothers, 2021-05-17:
## System
 - OS: Windows 10

## Describe the bug
When I ran UVtools today, Windows Defender popped up an alert and then deleted some of the files.  I downloaded a fresh copy and attempted to install it, but Defender deleted it during install.  Looks like a false positive to me, but you can never be too careful.

## To Reproduce
Steps to reproduce the behavior:
1. Download the latest MSI release
2. Run the installer

## Expected behavior
UVtools installs

## Screenshots
[image]

sn4k3, 2021-05-17:
Same as https://www.facebook.com/groups/uvtools/permalink/469791580797017/ ?

derekantrican, 2021-05-18:
https://github.com/AvaloniaUI/Avalonia/issues/5937

MattCarothers, 2021-05-28:
Looks like the Avalonia devs resolved this with Microsoft.   I was able to install and run 2.13 just now with no issues.

closed 2021-05-28 by MattCarothers