sn4k3/UVtools #205: [FEATURE REQUEST] Drain Hole in XYZ Accuracy Test

PrCRTL, 2021-05-24:
Hi, thanks for this great Tool.

I did the XYZ Accuracy Test today and missing a drainhole at the bottom... 
The print quality in this case a 40mm cube was according to that poor. 

best regards
Gerhard

sn4k3, 2021-05-24:
What do you mean? You can activate the drain hole by check the option: **Relief with a center hole** and **Hollow model**

PrCRTL, 2021-05-24:
A Drainhole goes through the Wall at the Bottom to let the inner resin pass through to avoid sucction.Tried all Options…Center hole makes a big hole in the base but it’s worthless if the part is direct on the Plate…Did I miss something?

sn4k3, 2021-05-24:
The hole at bottom is just a relief hole and to drain in case of hollow, but i can't see what's the problem you having. Can you send some pictures of the result or of the problem?

PrCRTL, 2021-05-24:
first picture showes how it comes generated with the relief hole and hollowed part... there is no way to drain anything as long the part sits on the plate...  You will see the bottom half of the print is not as it shoud be because of sucction....
on the second picture I made two examples in Black showing what I mean with a drain hole... as closest as possible to the plate to let the resin flow to remove sucction.... at least on one wall...
[image]
[image]

best regards
Gerhard

sn4k3, 2021-05-24:
> there is no way to drain anything as long the part sits on the plate...

Not really... The print prints upside down so it's always draining, the top is open, which means when plate moves up the resin drops on vat and not stay inside the piece. The center hole is just a relief it will drain with our without it.

Do not confuse drain hole with vent hole. What you are refering to and show on picture is a vent hole which the function is let air pass to relief the pressure. I may implement this in future as a improvement to your request.

The bottom layers problem you have are due the elephant foot and prolongated exposure, which you can ignore for this test.

However your model indicate problems on your printer, either mechanical or level or settings, i have seen some cubes by now and the most i see are just perfect, no artifacts on surface like you are getting, suction never been a problem on this small models, the cube i printed myself come out perfect and aligned walls.
Also the lack of text engravings mean you maybe overexposing the part which can also lead to some problems here

PrCRTL, 2021-05-24:
I see that you are using differnt terminology… in the Forums I’, in all are naming drain holes what you are calling Vent holes….

Don’t take me wrong, but you are simply wrong with your guess that its draining when the part is open… but this is not the right place to discuss that.

Finally I made my own cube with the holes and everything works as expected and looks perfect with the same settings… which are not perfect as you mentioned right… but that has its reason J

cheers

sn4k3, 2021-05-24:
Well i can be wrong, but to me the term drain seens more suited to remove something from inside, mostly a liquid. First entry on dictionary seens to agree with me, but i'm not an english native either so it can have other meanings that i'm not used to...
Glad that you able to fix yourself, i will add that feature in a future patch.

sn4k3, 2021-05-24:
Closed by mistake, reopen until implemented

sn4k3, 2021-05-29:
implemented

closed 2021-05-29 by sn4k3