sn4k3/UVtools #221: [FEATURE REQUEST] Ignore bottom layers setting in resin trap searching

djrampage, 2021-06-17:
In lychee slicer pro available raft pattern made my lines between supports and hollow area between lines.

[image]

Searching algorythm tests every hollow area of this pattern for resin trap and it takes about 3/4 time of detecting process.
Can you implement new setting for "Issues - Resin traps" like "Ignore raft layers count"? This option will reduce time waiting for computing issues.
On screenshot raft layes are 0-19.
Thank you.

sn4k3, 2021-06-17:
That raft pattern works good for you?

> 3/4 time of detecting process

Well only if you have a low layer count, 19 layers don't seen to time consuption... Have you test it? You can remove all layers but keep the 0 to 19 and run the process and see how much time is wasting there. 
UVtools can't actually know what is a raft, model or support, everything is a white pixel

djrampage, 2021-06-17:
Yes, this pattern works good, uses less amount of resin than solid raft and has no suction effect. Some creators from Patreon provides presupported files with that pattern so i need to solidify it or make something else for decrease detecting time.

Just compared detecting time on my next batch. 319 sec with lines pattern and 190 sec with solid raft.
20 layers @0.05mm thickness is default for this raft.

This pattern uses only a few creators so this is not critical.
I understand that UVTools can't identify that layers are raft so i suggested it as manual option.

I will not blame and understand if you consider this suggestion meaningless.

sn4k3, 2021-06-17:
I can put a option to ignore bottoms. How many bottom layers do you use?

djrampage, 2021-06-17:
Various slicers and rafts uses various various thickness. Usually it is 0,3-1,2mm (some presupported STLs made in Chitubox have really thick raft. Idk why).
For common 0.05mm layer thickness it is 6-24 layers.

sn4k3, 2021-06-17:
Can you send me a test file with that raft pattern so i can test?

djrampage, 2021-06-17:
Files for test. Just sliced file and file with fixed issues that i already printed:
https://drive.google.com/drive/folders/1ouut4EH3S2nfS4qQLo3Nm-ZK1UQqLTE8?usp=sharing

I measured some issues compute time difference on my PC (intel i5-7300u @2.6 ghz 2c/4t, 16gb ram, intel hd graphics 620, nvme ssd. Energy saving plan: performance)

Test with fresh sliced file where issues exists
| Issue type | With raft | Without raft | Diff |
| --- | --- | --- | --- |
| Islands/Overhangs | 145 sec | 132 sec | -13 sec |
| Resin traps | 425 sec | 216 sec | -209 sec |

Test file where where all issues have been fixed
| Issue type | With raft | Without raft | Diff |
| --- | --- | --- | --- |
| Islands/Overhangs | 132 sec | 127 sec | -5 sec | 
| Resin traps | 296 sec | 122 sec | -174 sec |

Yes, with a difference of 3/4 I got excited. Sorry :) Just about 50% only for resin traps.

sn4k3, 2021-06-17:
The resin trap would need a rewrite some day, but very hard to optimize when things are sequential dependent.
I will implement the ignore bottom layers setting. But the problem still happen to you because you have 6 bottom layers but raft goes up to layer 19. Which is a error, bottom layers should be raft size + 1 or 2 support layers to prevent failure. 
To a better comprehension of what raft settings shoud be read here: https://ameralabs.com/blog/default-3d-printing-raft-settings

To your reference on my PC it took to compute resin traps: (I9-9900k 32GB ram) is another beast ^^

```
#	Started	Time(s)	Description
6	17:52:12	166.18	Computing Issues
```

djrampage, 2021-06-17:
Oh, i see. But i using provided files and almost can't change raft thinkness. I see the way like move down whole model to decrease thickness of raft when i got .stl.
When files provided in slicer's project format i can edit premium parameters of supports and raft without premium account but not always i got project files like .lys.
I think i'll use described method as workaround.
Sometimes i need to do detect/fix serveral times that's because i posted this request.
Thank you for your work. Your tool is awesome and very unique!

sn4k3, 2021-06-17:
Pre fabricated files should be clean of issues...

djrampage, 2021-06-17:
Believe me, most of minis sculptors are hardly know how to make clean pre-supports and printing batches. I found tons of islands, overhangs and useless supports in pre-supported files of various sculptors.
I'm trying to promote your tool in patreon printing communities bc all issues became obvious after first detection and creators will make better quality of files with less amount of issues.

I showed issues to EdytheCross3D in his batch files and he was very impressed:
> Im not gonna lie, im not sure how long will it take for me to check my models like this. 
> But i would love to have this kind of workflow for our releases. Just to make sure Patrons can have the best quality of the model.
> <...>
> This is really helpful for everyone.
> 
> Tagging EllaArt @ellalala 
>    - Good read above. Something to consider! :slight_smile:

sn4k3, 2021-09-06:
Implemented

closed 2021-09-06 by sn4k3