sn4k3/UVtools #227: [BUG] Wrong icon for

FutureJames, 2021-06-22:
## Describe the bug
The icon used in the "Tools" menu for "Repair Layers and Issues" does not match the button with the same function in the "Issues" tab on the left side of the window.  There is a toolbox used in the "Tools" menu and a wrench used in the "Issues" tab.  These are the same function and should use the same button.  The wrench icon is used in the "Tools" menu for a completely different function ("Edit Print Parameters") 

## To Reproduce
Steps to reproduce the behavior:
1. Go to "Tools" menu
2. Read down to "Repair Layers and Issues"
4. See tool box icon
5. close menu
6. click on "Issues" tab
7. look at icon for "Repair Layers and Issues"
8. observe that it is a wrench 

## Expected behavior
A clear and concise description of what you expected to happen.

## Screenshots

[image]
[image]
[image]

## Additional context
The tool is not broken, the icons are just not being used consistently.  I assume that this is not intentional.

sn4k3, 2021-06-22:
Yeah, will update accordingly

FutureJames, 2021-06-23:
On that same topic, the icon for "Outline" and "Morph" are the same and also confusing.  (although they both have text labels, so less of an issue).  See attached screenshot
[image]

sn4k3, 2021-06-23:
Maybe i should change the **Outine** to **View**? Can't find a proper icon.
You can advice one or from: https://fontawesome.com/v5.15/icons?d=gallery&p=2&m=free

FutureJames, 2021-06-23:
I think "outline" makes a lot of sense as a label.  Perhaps one of these icons would be a good alternative:
https://fontawesome.com/v5.15/icons/vector-square?style=solid
https://fontawesome.com/v5.15/icons/square?style=regular

Also, maybe a fun one for "Elephant Foot" :)
https://fontawesome.com/v5.15/icons/republican?style=solid

sn4k3, 2021-06-23:
The normal square also look like an checkbox...
The vector is better but still not perfect.

The elephant foot icon is actually a elephant foot, but at it size don't seen like it

FutureJames, 2021-06-24:
Yeah that is a good point.  It does kind of look like a checkbox.
This one is probably closest to the concept of outlining an area:
https://fontawesome.com/v5.15/icons/draw-polygon?style=solid

This one also gives a bit of the feeling of outline multiple areas:
https://fontawesome.com/v5.15/icons/border-all?style=solid

sn4k3, 2021-06-25:
I think i will go with the square vector. Can always change latter ;)

closed 2021-06-25 by sn4k3