sn4k3/UVtools #232: Decode UVJ file: the case when JSON settings/config has an empty `Layers` array

Megidd, 2021-07-03:
### Discussed in https://github.com/sn4k3/UVtools/discussions/231

<div type='discussions-op-text'>

<sup>Originally posted by **Megidd** July  3, 2021</sup>
Sometimes I'm observing this exception, could it be a bug? The exception is observed *occasionally* 😕 

The troublemaker file is attached too.

[128034414.zip](https://github.com/sn4k3/UVtools/files/6758101/128034414.zip)

[image]
</div>

Megidd, 2021-07-03:
Also, this exception is thrown when `Layers` array is `null` rather than *empty*. So, maybe two checks might be needed:

1. Array is *not* empty
2. Array is *not* null

[image]

sn4k3, 2021-07-03:
> Also, this exception is thrown when `Layers` array is `null` rather than _empty_. So, maybe two checks might be needed:
> 
> 1. Array is _not_ empty
> 2. Array is _not_ null

Ok just need to put a **?** after layers: `JsonSettings.Layers?.Count`

sn4k3, 2021-07-07:
Fixed

closed 2021-07-07 by sn4k3