sn4k3/UVtools #233: [BUG] Application crashes when using any option from Tool or Calibration

Nume1977, 2021-07-04:
## System
 - OS: Windows 10 x64 v20H2
 - Slicer:  PrusaSlicer 2.3.1
 - Printer: ElfinStein (Mono Elfin 1)
 - OS freshly installed / UVTools 2.14 freshly installed (.msi version)

## Describe the bug
The program can open and convert file types from Prusa, but when using any menu function it will crash when starting it.

## To Reproduce
Steps to reproduce the behavior:
1. Go to 'Tool or Calibration option'
2. Click on 'Any option'
3. Start executing the tool or calibration selected
4. Program crashes

## Log
[errors.zip](https://github.com/sn4k3/UVtools/files/6759538/errors.zip)

sn4k3, 2021-07-04:
It's says: `Method not found: 'Void MessageBox.Avalonia.DTO.AbstractMessageBoxParams.set_MaxWidth(Double)'.`
Which can only missing if dll is missing or blocked.

Are you with windows defender? If yes try the following:

Uninstall UVtools
Disable Windows Defender
Install UVtools and retry

Does it works now?

Nume1977, 2021-07-04:
Followed your instructions, installed UVTools with AV disabled.
It worked. 
Strange thing is there are no entries in the AV logs of any file being blocked on the first install ...

sn4k3, 2021-07-04:
Also here: https://github.com/sn4k3/UVtools/issues/213

closed 2021-07-04 by sn4k3