sn4k3/UVtools #234: [BUG]

Judoguy12, 2021-07-04:
<span style="color:red">
**! Before continue, please look/search on closed topics if your case was already been discussed, if yes and related, comment there instead.**
**! Fill the report template and remove the sections you do not used on this bug report**
</span>

## System
 - OS: Fedora 34
 - Slicer: Prusa Slicer 
 - Printer: Photon Mono

## Describe the bug
Getting no pad can be generated error in Prusa slicer when adding supports

## To Reproduce
Steps to reproduce the behavior:
1. Go to prusa slicer
2. Click on auto generate supports
3. See error

## Expected behavior
Pad to be generated for supports to sit on
## Screenshots
[image]

## Additional context
It works with normal profiles

sn4k3, 2021-07-04:
Thats a PrusaSlicer related issue, not UVtools.
But my guess would be the lack of space on plate? Try to scale model by 50% just to see if is that the problem

Judoguy12, 2021-07-04:
Interesting it worked at 0.05 profile, yeah that's probably ps based

Sent from ProtonMail mobile

-------- Original Message --------

sn4k3, 2021-07-04:
Pads should not be larger than build plate, but you can trim them with settings

closed 2021-07-05 by sn4k3