sn4k3/UVtools #236: [BUG] UVtools 2.14.1 crashes on launch

codehearted, 2021-07-08:
## System
 - OS: macOS Big Sur 11.3.1 (Running on M1 MacBook Air - So UV Tools is running in Rosetta 2 Emulation Mode)
 - Slicer: LycheeSlicer 3.4.0, but it doesn't matter for this bug.
 - Printer: Anycubic Mono X and Anycubic Photon (Original), but this is also irrelevant.

## Describe the bug
I had been using UVtools 2.13.4 without issue for a while now, but was hoping for a feature that would allow me to delete layers at the bottom to remove islands just slightly above the build plate, so figured I would roll the dice and upgrade when I noticed that 2.14.1 was available today. I used the "automatic upgrade" feature, which seemed rather manual since it just opened the web page to download the new release, downloaded 2.14.1, replaced the old one, and opened it, and it crashed on open. Tried it again, crashed again. Every single time. Right clicking to open changed nothing. I tried the old version, it still works fine. So I'm going back to the old version.

## To Reproduce
Steps to reproduce the behavior:
1. Download 2.14.1 from https://github.com/sn4k3/UVtools/releases/download/v2.14.1/UVtools_osx-x64_v2.14.1.zip
2. Uncompress it, optionally move it to Applications if you are optimistic
3. Run it
4. It crashes

## Expected behavior
It does not crash and runs as normal

## Screenshots
It's just your standard "Problem Report for Avalonia Application" for the title, and "Avalonia Application quit unexpectedly." below that, "Click Reopen to open the application again. This report will be sent to Apple automatically."
Really not worth a screenshot. There is a lot of info in the crash log, which would not be captured adequately in a screenshot in any case, which I'll paste below.

## Files
Not Applicable.
 
## Exceptions Log file
I'm not sure where this AppData folder would be, I didn't see it inside the UVtools application package folder or in my user/Library/Application Support folder (no UVtools folder there) or the system Application Support. I'm guessing it didn't get launched enough to write anything anyway, but I'd be interested to know where exactly that log gets written on a Mac, if someone knows.

However there is a large crash log as I mentioned above, generated by the OS. I'll put that in Additional Context since no one would want to try to scroll past it to read anything else.

## Additional context
Here's that crash log:

'''

Process:               UVtools [90522]
Path:                  /Applications/UVtools.app/Contents/MacOS/UVtools
Identifier:            com.UVtools
Version:               2.14.1 (2.14.1)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           UVtools [90522]
User ID:               501

Date/Time:             2021-07-08 01:47:33.496 -0700
OS Version:            macOS 11.3.1 (20E241)
Report Version:        12
Anonymous UUID:        4BEF40E0-1FAF-F4B0-B79E-2D831BBC5207

Sleep/Wake UUID:       989ABA2E-E6CA-4913-8322-40FF91F41E12

Time Awake Since Boot: 890000 seconds
Time Since Wake:       49000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	0x00007ffe9458aab8 ???
1   libsystem_kernel.dylib        	0x00007fff2033f946 __pthread_kill + 10
2   libsystem_c.dylib             	0x00007fff202c3411 abort + 120
3   libcoreclr.dylib              	0x000000010cde590e PROCAbort + 30
[... 562 lines cut ...]
Bluetooth: Version 8.0.4d18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: pci1b21,1142, sppci_usbxhci, Thunderbolt@9,0,0
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: USB3.0 Hub
USB Device: USB 3.1 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB Storage
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: Thunderbolt Dock
USB Device: USB 3.0 Bus
USB Device: ASM107x
USB Device: Backup+ Hub
USB Device: Expansion Desk
USB Device: Backup+ Hub BK
USB Device: WEB CAMERA M9 Pro
USB Device: TUSB3410 Boot Device
USB Device: ASM107x
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Device: Thunderbolt Dock, Other World Computing, 1, 27.2

'''

codehearted, 2021-07-08:
I just went ahead and tested 2.14.0 to see if it had the same issue. It does not. It's only been a few minutes, but so far 2.14.0 seems to work as well as 2.13.4 did. I'm switching to 2.14.0 for now instead.

sn4k3, 2021-07-08:
2.14.1 upgraded opencv which introduced some new stuff that may not compatible with emulation, from the logs you sent i can't see it loaded so maybe is that. To access logs, go to UVtools -> Help -> Open settings folder. Is there a errors.log?

The best way is to native compile to M1, as you have one maybe you can help me compiling the opencv library required to run?

sn4k3, 2021-07-08:
Lets try this by multiple trys:

1. Download: https://we.tl/t-mhVUQVBFFu
1. Extract folder on dekstop
1. Open the folder and copy **libcvextern.dylib**
1. Go to UVtools v2.14.1 app file
2. Right click
3. Show package contents
4. Navigate to: Contents/MacOS
5. Paste file there and replace

Try to run

If fail try to:
1. Copy the **libusb-1.0.0.dylib** from step 2. into step 4.
1. Re-run

If it fail try to:
1. Copy the all the contents from folder from step 2. into step 4.
1. Re-run

rfield19, 2021-07-08:
I had exactly the same issue as OP with v2.14.1.
Followed the steps outlined above - first the step with libcvextern.dylib, and then the following two steps when that didn't work.  No difference though - still crashes upon opening.
I'm running it on an M1 machine by the way - noticed OP was running on an M1 mac as well, so maybe relevant.

sn4k3, 2021-07-08:
My build v2.14.0 was compiled on Monjave x64 and v2.14.1 was compiled on Catalina x64... But my guess is something new on the opencv that is missing...

Try put the x64/libcvextern.dylib into UVtools/Contents/MacOS
Should not fix but worth the try

rfield19, 2021-07-08:
Tried the x64/libcvextern.dylib - no difference unfortunately, still crashes on opening!

sn4k3, 2021-07-08:
Ok i compile it under Monjave and send the libs to test out

sn4k3, 2021-07-08:
I also have the bigsur libs.
Try same process with this files: https://we.tl/t-gnRgLCiU8K

rfield19, 2021-07-08:
Sorry, no luck!  Tested with - from bigsur archive
1) libcvextern.dylib (not working)
2) libusb-1.0.0.dylib (not working)
3) libcvextern.dylib from x64 dir (not working)

rfield19, 2021-07-08:
ok, so did a bit of digging...  
Went through all the files that had changed since 12.14.0 and stuck them into 12.14.1 one by one.
If I took the libcvextern.dylib from 12.14.0 release and put it into the 12.14.1 executable then everything works correctly...

The libcvextern in the 12.14.0 release is quite a bit smaller than the ones you've linked to above, so maybe it's much older?
It's dated 5th March and is 43,672,084 bytes (sorry, not sure how to find out the version of a dylib!)

sn4k3, 2021-07-08:
libcvextern.dylib:
v2.14.0  = OpenCV 4.5.1
v2.14.1  = OpenCV 4.5.2

Is some dependency that changed on opencv452 that mac refuse to load or have in compability mode.
It will be hard to know what is.

While old library works with new uvtools, it can give problems on flip operations

rfield19, 2021-07-08:
Just throwing it out there that I'm happy to test any native m1 version you experiment with - that might 100% fix the problem :)

sn4k3, 2021-07-08:
Native UVtools on M1 will require .NET 6.0: https://github.com/sn4k3/UVtools/issues/187

rfield19, 2021-07-08:
Ah ok, that's a shame!  I was going to offer to compile native m1 opencv4.5.2 for you, but guess that won't help at all if it needs .net v6 :/

sn4k3, 2021-07-08:
Yeah, but will be welcome when i upgrade to .NET 6.0

codehearted, 2021-07-08:
Just tried your first set of solutions and got the same results as rfield19 (from the https://we.tl/t-mhVUQVBFFu) libraries, and decided it is a safe assumption I would get the same results on the second set (the Big Sur libraries at https://we.tl/t-gnRgLCiU8K -- please let me know if you'd like a second set of tests on those) so I haven't tried them yet. I did go ahead and try the copying libcvextern (only) from 2.14.0 to 2.14.1 as rfield19 said worked for them, and it worked for me as well. Would that lose any functionality?

Also, although the Settings menu item in File works fine, "Open Settings Folder" in the Help menu does not seem to do anything. I hope that helps.

sn4k3, 2021-07-08:
> Would that lose any functionality?

Only if you start to play with Flip function... Other than that is safe

sn4k3, 2021-07-08:
Try this monjave build.

Remove old things you put on uvtools before.
Start by x64 folder dylib into uvtools contents

https://we.tl/t-fpaE3V9JbW

rfield19, 2021-07-08:
No difference with new files unfortunately - doesn't load.
With a fresh download of 2.14.1 I tried the dylib in the x64 dir, the one in the root dir, and then all files from the root dir.

sn4k3, 2021-07-08:
with libusb included right?
Is there a way to run terminal in emulated way? x64?

rfield19, 2021-07-08:
that was with libusb included also, yes.
I'm not aware of any way of running an emulated terminal apart from inside a vm, but that's probably not a good option!

sn4k3, 2021-07-08:
I mean, run terminal/console in Rosetta 2 Emulation Mode

rfield19, 2021-07-08:
Perhaps an old version of iterm might run in rosetta2 mode - not sure.

sn4k3, 2021-07-08:
Lets try natively to see.

Open a terminal on UVtools content folder, and try these commands:

```
otool -L libcvextern.dylib
otool -L UVtools
```

rfield19, 2021-07-08:
OK, I've downloaded an old version of iterm2 - v 3.3.1.2 and it seems to be running under rosetta2 (I've got an app called silicon info running in the taskbar that tells me what it's running as).

If I type 
otool -L /Applications/UVtools.app/Contents/MacOS/libcvextern.dylib
then (same output on both arm and x64 terminals) - I get

/Applications/UVtools.app/Contents/MacOS/libcvextern.dylib (architecture x86_64):
	@rpath/libcvextern.dylib (compatibility version 0.0.0, current version 0.0.0)
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/System/Library/Frameworks/AVFoundation.framework/Versions/A/AVFoundation (compatibility version 1.0.0, current version 2.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1265.9.0)
	/System/Library/Frameworks/CoreMedia.framework/Versions/A/CoreMedia (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo (compatibility version 1.2.0, current version 1.5.0)
	/System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore (compatibility version 1.2.0, current version 1.11.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.250.1)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1671.60.107)
	/System/Library/Frameworks/OpenCL.framework/Versions/A/OpenCL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.4)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1575.17.0)
	/System/Library/Frameworks/CoreText.framework/Versions/A/CoreText (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1575.17.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)

rfield19, 2021-07-08:
and if I do
otool -L /Applications/UVtools.app/
then I get:

error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/otool-classic: can't map file: /Applications/UVtools.app (Invalid argument)

sn4k3, 2021-07-08:
> and if I do
> otool -L /Applications/UVtools.app/
> then I get:
> 
> error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/otool-classic: can't map file: /Applications/UVtools.app (Invalid argument)

Yes, must be: otool -L /Applications/UVtools.app/Contents/MacOS/UVtools

Can you run the otool on the working libcvextern.dylib (from previous version)

rfield19, 2021-07-09:
OK, that's better!

otool -L /Applications/UVtools.app/Contents/MacOS/UVtools

/Applications/UVtools.app/Contents/MacOS/UVtools:
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 800.7.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.0.0)

on the default 2.14.1 app )no changes) I get

otool -L /Applications/UVtools.app/Contents/MacOS/libcvextern.dylib
@rpath/libcvextern.dylib (compatibility version 0.0.0, current version 0.0.0)
	**/usr/local/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)**
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/System/Library/Frameworks/AVFoundation.framework/Versions/A/AVFoundation (compatibility version 1.0.0, current version 2.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1463.2.1)
	/System/Library/Frameworks/CoreMedia.framework/Versions/A/CoreMedia (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo (compatibility version 1.2.0, current version 1.5.0)
	/System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore (compatibility version 1.2.0, current version 1.11.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.60.1)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 2022.20.117)
	/usr/local/opt/tesseract/lib/libtesseract.4.dylib (compatibility version 5.0.0, current version 5.1.0)
	/usr/local/opt/leptonica/lib/liblept.5.dylib (compatibility version 6.0.0, current version 6.4.0)
	/System/Library/Frameworks/OpenCL.framework/Versions/A/OpenCL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 904.4.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1770.255.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1770.255.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)

On the 2.14.1 app but with the 2.14.0 libcvextern.dylib copied into it, I get:

/Applications/UVtools.app/Contents/MacOS/libcvextern.dylib **(architecture x86_64):**
	@rpath/libcvextern.dylib (compatibility version 0.0.0, current version 0.0.0)
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/System/Library/Frameworks/AVFoundation.framework/Versions/A/AVFoundation (compatibility version 1.0.0, current version 2.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1265.9.0)
	/System/Library/Frameworks/CoreMedia.framework/Versions/A/CoreMedia (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo (compatibility version 1.2.0, current version 1.5.0)
	/System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore (compatibility version 1.2.0, current version 1.11.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.250.1)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1671.60.107)
	/System/Library/Frameworks/OpenCL.framework/Versions/A/OpenCL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.4)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1575.17.0)
	/System/Library/Frameworks/CoreText.framework/Versions/A/CoreText (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1575.17.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)

sn4k3, 2021-07-09:
Ok so the diffs are:

[image]

Can you confirm if such blue highlight libraries exists on your mac?

rfield19, 2021-07-09:
The only one that is present from the blue highlighted entries is line 17 on 2.14.0
(the coretext entry).

The others don't exist.  
For 2.14.1 there's no /usr/local/opt directory on the mac at all, just /usr/local

sn4k3, 2021-07-09:
Ok try another command:

`objdump -r --dylibs-used /Applications/UVtools.app/Contents/MacOS/UVtools`

rfield19, 2021-07-09:
objdump -r --dylibs-used /Applications/UVtools.app/Contents/MacOS/UVtools

/Applications/UVtools.app/Contents/MacOS/UVtools:	file format mach-o 64-bit x86-64

sn4k3, 2021-07-09:
`objdump -r --dylibs-used /Applications/UVtools.app/Contents/MacOS/libcvextern.dylib` ?

rfield19, 2021-07-09:
/Applications/UVtools.app/Contents/MacOS/libcvextern.dylib:
	@rpath/libcvextern.dylib (compatibility version 0.0.0, current version 0.0.0)
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/System/Library/Frameworks/AVFoundation.framework/Versions/A/AVFoundation (compatibility version 1.0.0, current version 2.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1265.9.0)
	/System/Library/Frameworks/CoreMedia.framework/Versions/A/CoreMedia (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo (compatibility version 1.2.0, current version 1.5.0)
	/System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore (compatibility version 1.2.0, current version 1.11.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.250.1)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1671.60.107)
	/System/Library/Frameworks/OpenCL.framework/Versions/A/OpenCL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.4)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1575.17.0)
	/System/Library/Frameworks/CoreText.framework/Versions/A/CoreText (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1575.17.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)

rfield19, 2021-07-09:
Oh, and if I do the same command on the default 2.14.1 app, unmodified, then I only get:

objdump -r --dylibs-used /Applications/UVtools.app/Contents/MacOS/libcvextern.dylib

/Applications/UVtools.app/Contents/MacOS/libcvextern.dylib:	file format mach-o 64-bit x86-64

sn4k3, 2021-07-09:
Getting hard to debug :(
Problem with otool, i think it always shows dependencies, whatever they are or not in the system. I need to dig into this...

rfield19, 2021-07-09:
Sure, happy to try anything at this end if it helps.  I'm not a good programmer, but I'm enthusiastic !

sn4k3, 2021-07-09:
In the end i think what is missing is the 

libtesseract.4.dylib
liblept.5.dylib

as libusb is present on the build, but libtesseract and liblep not...

milescsmith, 2021-07-13:
I'd just like to chime in here that I'm running an macOS Catalina (10.15.7) with an Intel Mac and any version above 2.14.0 instantly crashes for me too.

RWH70, 2021-07-13:
Confirming the problem on MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports) macOS Big Sur 11.4
Latest working version is UVtools 2.14.0
All subsequent versions 2.14.1 - 2.14.3 crash at startup by mistake.

rfield19, 2021-07-13:
Just for info, the same earlier 'band-aid' solution works for 2.14.3 ; if one copies the libcvextern.dylib from 2.14.0 into the new 2.14.3 version then it runs.

sn4k3, 2021-07-13:
I was able to compile it without the libtesseract.
Please try by replace only this libcvextern.dylib: https://we.tl/t-1YCIcYlXRu

rfield19, 2021-07-13:
> I was able to compile it without the libtesseract.
> Please try by replace only this libcvextern.dylib: https://we.tl/t-1YCIcYlXRu

Sorry, copied in the new dylib and same result  - crashes while loading.
Let me know if you want anything testing whilst the new file is in place.

sn4k3, 2021-07-13:
> Sorry, copied in the new dylib and same result - crashes while loading.
> Let me know if you want anything testing whilst the new file is in place.

Dam... Tried with libusb also?

rfield19, 2021-07-13:
Yes, first attempt just with libcvextern, second attempt with both libcvextern and libusb.
Must be really frustrating :/

rfield19, 2021-07-13:
I've got a new bit of info for you though - might help!

This is what I get when using the new dylibs with
objdump -r --dylibs-used /Applications/UVtools.app/Contents/MacOS/libcvextern.dylib

**objc[76371]: Class AMSupportURLConnectionDelegate is implemented in both /usr/lib/libauthinstall.dylib (0x1eee6ec10) and /System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x117fbc2b8). One of the two will be used. Which one is undefined.
objc[76371]: Class AMSupportURLSession is implemented in both /usr/lib/libauthinstall.dylib (0x1eee6ec60) and /System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x117fbc308). One of the two will be used. Which one is undefined.**

/Applications/UVtools.app/Contents/MacOS/libcvextern.dylib:
	@rpath/libcvextern.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/System/Library/Frameworks/AVFoundation.framework/Versions/A/AVFoundation (compatibility version 1.0.0, current version 2.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1265.9.0)
	/System/Library/Frameworks/CoreMedia.framework/Versions/A/CoreMedia (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo (compatibility version 1.2.0, current version 1.5.0)
	/System/Library/Frameworks/QuartzCore.framework/Versions/A/QuartzCore (compatibility version 1.2.0, current version 1.11.0)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.250.1)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 1671.60.107)
	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
	/System/Library/Frameworks/OpenCL.framework/Versions/A/OpenCL (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.4)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1575.17.0)
	/System/Library/Frameworks/CoreText.framework/Versions/A/CoreText (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1575.17.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)

sn4k3, 2021-07-13:
can you try download the osx legacy package and try there instead?

rfield19, 2021-07-13:
How do I run the legacy package?  It's just a zip file that unzips to a directory rather than an app.

If I run <dir>/UVtools exec then it just whinges about 'developer cannot be verified' for every single file it tries to access!

sn4k3, 2021-07-13:
Dam even worse xD

sn4k3, 2021-07-13:
I think after install libusb it worked for me...

`brew install libusb`

rfield19, 2021-07-13:
I've installed libusb from brew, and not seeing any difference here - still crashes on launch.
Which combination worked after you installed brew/libusb - the original 2.14.3 one or something else?

rfield19, 2021-07-13:
After installing brew / libusb however, I no longer get the 

_"objc[76371]: Class AMSupportURLConnectionDelegate is implemented in both /usr/lib/libauthinstall.dylib (0x1eee6ec10) and /System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x117fbc2b8). One of the two will be used. Which one is undefined.
objc[76371]: Class AMSupportURLSession is implemented in both /usr/lib/libauthinstall.dylib (0x1eee6ec60) and /System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x117fbc308). One of the two will be used. Which one is undefined."_

error, so that seems a step forward.

sn4k3, 2021-07-13:
> I've installed libusb from brew, and not seeing any difference here - still crashes on launch.
> Which combination worked after you installed brew/libusb - the original 2.14.3 one or something else?

I downloaded latest UVtools.
Put the new libcvextern.dylib

And installed this:

`brew install libusb libjpeg libpng libgeotiff`

From a fresh Catalina VM mac

rfield19, 2021-07-13:
That's still not working here unfortunately.  Followed the same steps 
ie
brew install libusb libjpeg libpng libgeotiff
then copied the new libcvextern.dylib into the 2.14.3 app.  Still crashing on launch.

Do you have a Big Sur (11.4) VM rather than Catalina (11.3)?
Would also be worthwhile having someone try the same procedure on a non M1 mac perhaps, in case it's a rosetta2 issue?

sn4k3, 2021-07-13:
my VM is x64.
Try on your other mac please

rfield19, 2021-07-13:
OK, some good news!
On my x64 mac (running big sur) I installed the brew components and then copied in the replacement libcvextern.dylib ... and it works - well done :)

Still not working on m1 mac though, so maybe a rosetta issue?

sn4k3, 2021-07-13:
Ok, so i guess the only required dependency for x64 is libusb now.
For roseta something is missing, maybe also the libusb for arm?

rfield19, 2021-07-13:
As I say, I'm not a programmer so this is just a guess, but I think brew libusb is already arm64?
Looking at the libusb code here
https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb

it shows:

 bottle do
    sha256 cellar: :any,                 **arm64_big_sur:** "1c40f64450705461a5373c3d54257e646d39914d44bffaf9d957bbe063db2129"
    sha256 cellar: :any,                 big_sur:       "74e0067e968ddbea31e070885ae86bc1db5c66fd157588e84576e653e62894c8"
    sha256 cellar: :any,                 catalina:      "034ae259f17afb5894860cdb1786fd6d391359e8d221c0f765eceed6210b60df"
    sha256 cellar: :any,                 mojave:        "1318e1155192bdaf7d159562849ee8f73cb0f59b0cb77c142f8be99056ba9d9e"
    sha256 cellar: :any_skip_relocation, x86_64_linux:  "8e3ee4fb54d2e1150c126c6d696b887e1422961d9fe6d18b26c7a4d5a2313ea7"
  end

sn4k3, 2021-07-13:
libusb is the only diference now.
The only way to fix this is try to get rid of it from libcvextern.

sn4k3, 2021-07-16:
New release will work better on x64, but still need the `brew install libusb` 
Tell me the result, and also if still not working on M1

[37 more replies]

closed 2021-08-02 by sn4k3