sn4k3/UVtools #239: [BUG] Error generated with exposure time finder

Martichoke, 2021-07-10:
## System
 - OS: Windows 10
 - Slicer: Chitubox, Photon Workshop
 - Printer: Mono X

## Describe the bug
Please check attached document.

sn4k3, 2021-07-10:
Will need the sliced file and settings used to replicate

Martichoke, 2021-07-10:
Please find attached.
Also, in the fourth picture, you can notice I am scrolled to the very bottom of the Exposure time finder pop-open screen. The Pattern loaded model drop-down is not available.
If I generate the model and then reopen settings, it becomes available like in the fifth picture.
Thank you.

https://drive.google.com/file/d/1e1WO0OjtkI3lEPQWwt5Bvv8K3YKP_y5s/view?usp=sharing

[image]
[image]
[image]
[image]
[image]

sn4k3, 2021-07-10:
Ah you are doing it wrong...
You must have 1 sliced body, and that file is already full with 8 bodies... That will never work.
Also never call your file R_E_R_F when printing

sn4k3, 2021-07-10:
Do the following steps to keep only one object on plate:

1. Go to layer 0
2. Hold shift + right click on a white base
3. Go to tools -> Pixel arithmetic
4. Select Keep Region and Apply

Then you can use Multiple exposures

sn4k3, 2021-07-10:
Also keep in mind that test will be slow to execute, either wait for next version of UVtools which almost eliminate lifts OR use multiple brightness instead which print all objects at once

sn4k3, 2021-07-11:
fixed

closed 2021-07-11 by sn4k3

Martichoke, 2021-07-11:
Thanks. That worked great. I just assumed it would delete existing and resize to suit.

sn4k3, 2021-07-11:
> Thanks. That worked great. I just assumed it would delete existing and resize to suit.

When using the 'Pattern loaded model' it does exactly that, pattern whatever you have currently loaded. For example you slice a amarelabs tower and use that feature.