sn4k3/UVtools #240: [BUG]

Calvir, 2021-07-14:
## System
 - OS: Windows 10 x64 v20H2
 - Slicer: chitubox
 - Printer: Halot-one

## Describe the bug
A clear and concise description of what the bug is.
the converted file to cxdlp which was working before, don't work anymore. I did a firmware update of my printer yesterday, and all the files I converted with UVtools don't worked now. I think creality modified is file format.
[test slice.zip](https://github.com/sn4k3/UVtools/files/6813865/test.slice.zip)

## To Reproduce
Steps to reproduce the behavior:
1. slice a file with chitubox in cbddlp file format
2. convert in cxdlp with UVtolls

## Expected behavior

## Files

 Thank you for you work and updates

sn4k3, 2021-07-14:
Posting header differenteces.
original left, uvtools right:

[image]
[image]

sn4k3, 2021-07-16:
fixed

closed 2021-07-16 by sn4k3

Calvir, 2021-07-25:
after test, it work fine.
Thank you