sn4k3/UVtools #243: UVtools 2.15.0 Crashes and Fails to Start

catzpaw164, 2021-07-20:
Running macOS Catalina version 10.15.7
iMac (Retina 4K 21.5 inch, 2019)
Processor: 3.6 GHz Quad-Core Intel Core i3
Memory: 16 GB 2400 MHz DDR4
Graphics: Radeon Pro 555X 2 GB

Error Log as follows:

Process:               UVtools [4501]
Path:                  /Applications/UVtools.app/Contents/MacOS/UVtools
Identifier:            com.UVtools
Version:               2.15.0 (2.15.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           UVtools [4501]
User ID:               501

Date/Time:             2021-07-20 14:49:36.070 -0400
OS Version:            Mac OS X 10.15.7 (19H1217)
Report Version:        12
Anonymous UUID:        85483540-4338-75B8-3620-372A13073CCA

Sleep/Wake UUID:       5E9261AE-B72D-4519-9A89-2FF25E14D521

Time Awake Since Boot: 78000 seconds
Time Since Wake:       1200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6b89e33a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff6b95ae60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff6b825808 abort + 120
3   libcoreclr.dylib              	0x0000000101d2690e PROCAbort + 30
4   libcoreclr.dylib              	0x00000001018f9657 PROCEndProcess(void*, unsigned int, int) + 215
5   libcoreclr.dylib              	0x0000000101d293bc UnwindManagedExceptionPass1(PAL_SEHException&, _CONTEXT*) (.cold.1) + 44
6   libcoreclr.dylib              	0x0000000101b489a6 UnwindManagedExceptionPass1(PAL_SEHException&, _CONTEXT*) + 822
7   libcoreclr.dylib              	0x0000000101b48a36 DispatchManagedException(PAL_SEHException&, bool) + 134
8   libcoreclr.dylib              	0x0000000101aac2a4 IL_Throw(Object*) + 548
9   ???                           	0x00000001102fb8bd 0 + 4566530237
10  ???                           	0x000000011242d217 0 + 4601336343
11  ???                           	0x000000011242cccd 0 + 4601334989
12  ???                           	0x0000000110c737c9 0 + 4576458697
13  ???                           	0x0000000110738704 0 + 4570973956
14  libcoreclr.dylib              	0x0000000101be93c9 CallDescrWorkerInternal + 124
15  libcoreclr.dylib              	0x0000000101a3debf MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 1519
16  libcoreclr.dylib              	0x000000010191958a RunMain(MethodDesc*, short, int*, PtrArray**) + 746
17  libcoreclr.dylib              	0x00000001019198c3 Assembly::ExecuteMainMethod(PtrArray**, int) + 387
18  libcoreclr.dylib              	0x000000010195640d CorHost2::ExecuteAssembly(unsigned int, char16_t const*, int, char16_t const**, unsigned int*) + 509
19  libcoreclr.dylib              	0x0000000101902df2 coreclr_execute_assembly + 242
20  libhostpolicy.dylib           	0x0000000101869bd3 run_app_for_context(hostpolicy_context_t const&, int, char const**) + 1443
21  libhostpolicy.dylib           	0x000000010186aeda corehost_main + 234
22  libhostfxr.dylib              	0x0000000101805a11 fx_muxer_t::handle_exec_host_command(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, host_startup_info_t const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::unordered_map<known_options, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >, known_options_hash, std::__1::equal_to<known_options>, std::__1::allocator<std::__1::pair<known_options const, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > > > const&, int, char const**, int, host_mode_t, char*, int, int*) + 1281
23  libhostfxr.dylib              	0x0000000101804b63 fx_muxer_t::execute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, int, char const**, host_startup_info_t const&, char*, int, int*) + 611
24  libhostfxr.dylib              	0x000000010180250a hostfxr_main_startupinfo + 138
25  com.UVtools                   	0x00000001017ce823 exe_start(int, char const**) + 1795
26  com.UVtools                   	0x00000001017cea6f main + 143
27  libdyld.dylib                 	0x00007fff6b756cc9 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff6b897dfa mach_msg_trap + 10
[... 473 lines cut ...]
__LINKEDIT                       391.6M       15 
__OBJC_RO                         32.3M        1 
__OBJC_RW                         1908K        2 
__TEXT                           191.0M      284 
__UNICODE                          564K        1 
mapped file                       89.8M      175 
shared memory                      632K       14 
===========                     =======  ======= 
TOTAL                              3.6G     3905 
TOTAL, minus reserved VM space     3.2G     3905 

Model: iMac19,2, BootROM 1554.120.15.0.0, 4 processors, Quad-Core Intel Core i3, 3.6 GHz, 16 GB, SMC 2.47f3
Graphics: kHW_AMDRadeonPro555XItem, Radeon Pro 555X, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E3
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E3
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Oct 28 2020 13:06:41 version 9.30.357.42.32.5.48 FWID 01-4f9e2d7c
Bluetooth: Version 7.0.6f8, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD HTS541010A9E632, 1 TB
USB Device: USB 3.1 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: USB3.0 Hub
USB Device: Mass Storage Device
Thunderbolt Bus: iMac, Apple Inc., 47.4

sn4k3, 2021-07-20:
Duplicate of #236
short answer: you need install a dependency: `brew install libusb` on a terminal

catzpaw164, 2021-07-20:
Do I do this opening terminal entering that command? Or is there another process?

sn4k3, 2021-07-20:
> Do I do this opening terminal entering that command? Or is there another process?

Yes. You need homebrew installed before enter that command

catzpaw164, 2021-07-20:
Thank you for your instructions. This has fixed the problem.

closed 2021-07-20 by catzpaw164