sn4k3/UVtools #244: v.2.15.1 OSX crash on opening

samuelemarazzita, 2021-07-24:
## System
 - OS: MacOS Big Sur 11.5

## Describe the bug
When I try to open the application, it simply crash and ask me to report or to open again the application (that then crash again).

PaleHaunt, 2021-07-24:
I'm having the same issue on my Mac using Catalina as well. Crash on opening over and over.

sn4k3, 2021-07-24:
Mac is x64 or arm M1?

samuelemarazzita, 2021-07-24:
X64

PaleHaunt, 2021-07-24:
> Mac is x64 or arm M1?

x64 as well

sn4k3, 2021-07-24:
Strange... Try to install libusb and relaunch UVtools.
brew install libusb
See previous issues there is same topic open before.

samuelemarazzita, 2021-07-25:
Doesn't fix the issue. I will continue using the previous version.

PaleHaunt, 2021-07-25:
Same here. I installed the libusb upgrade, same crash on launch.

PaleHaunt, 2021-07-25:
Process:               UVtools [17838]
Path:                  /Applications/UVtools.app/Contents/MacOS/UVtools
Identifier:            com.UVtools
Version:               2.15.1 (2.15.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           UVtools [17838]
User ID:               501

Date/Time:             2021-07-25 10:23:37.126 -0500
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Bridge OS Version:     4.6 (17P6610)
Anonymous UUID:        2CB40DF4-7A06-E91E-27D5-005C5B3B7275

Time Awake Since Boot: 67000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6796d33a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff67a29e60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff678f4808 abort + 120
3   libcoreclr.dylib              	0x0000000106cf48ae PROCAbort + 30
4   libcoreclr.dylib              	0x00000001068c75f7 PROCEndProcess(void*, unsigned int, int) + 215
5   libcoreclr.dylib              	0x0000000106cf735c UnwindManagedExceptionPass1(PAL_SEHException&, _CONTEXT*) (.cold.1) + 44
6   libcoreclr.dylib              	0x0000000106b16946 UnwindManagedExceptionPass1(PAL_SEHException&, _CONTEXT*) + 822
7   libcoreclr.dylib              	0x0000000106b169d6 DispatchManagedException(PAL_SEHException&, bool) + 134
8   libcoreclr.dylib              	0x0000000106a7a244 IL_Throw(Object*) + 548
9   ???                           	0x000000011d2bb8fd 0 + 4784371965
10  ???                           	0x000000011f3fc537 0 + 4819240247
11  ???                           	0x000000011f3fbfed 0 + 4819238893
12  ???                           	0x000000011dc336a9 0 + 4794300073
13  ???                           	0x000000011d6f8704 0 + 4788815620
14  libcoreclr.dylib              	0x0000000106bb7369 CallDescrWorkerInternal + 124
15  libcoreclr.dylib              	0x0000000106a0be5f MethodDescCallSite::CallTargetWorker(unsigned long const*, unsigned long*, int) + 1519
16  libcoreclr.dylib              	0x00000001068e752a RunMain(MethodDesc*, short, int*, PtrArray**) + 746
17  libcoreclr.dylib              	0x00000001068e7863 Assembly::ExecuteMainMethod(PtrArray**, int) + 387
18  libcoreclr.dylib              	0x00000001069243ad CorHost2::ExecuteAssembly(unsigned int, char16_t const*, int, char16_t const**, unsigned int*) + 509
19  libcoreclr.dylib              	0x00000001068d0d92 coreclr_execute_assembly + 242
20  libhostpolicy.dylib           	0x0000000106837bd3 run_app_for_context(hostpolicy_context_t const&, int, char const**) + 1443
21  libhostpolicy.dylib           	0x0000000106838eda corehost_main + 234
22  libhostfxr.dylib              	0x00000001067d3a11 fx_muxer_t::handle_exec_host_command(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, host_startup_info_t const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::unordered_map<known_options, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >, known_options_hash, std::__1::equal_to<known_options>, std::__1::allocator<std::__1::pair<known_options const, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > > > const&, int, char const**, int, host_mode_t, char*, int, int*) + 1281
23  libhostfxr.dylib              	0x00000001067d2b63 fx_muxer_t::execute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, int, char const**, host_startup_info_t const&, char*, int, int*) + 611
24  libhostfxr.dylib              	0x00000001067d050a hostfxr_main_startupinfo + 138
25  com.UVtools                   	0x0000000106792823 exe_start(int, char const**) + 1795
26  com.UVtools                   	0x0000000106792a6f main + 143
27  libdyld.dylib                 	0x00007fff67825cc9 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff67966dfa mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff67967170 mach_msg + 60
2   libcoreclr.dylib              	0x00000001068cddc8 MachMessage::Receive(unsigned int) + 72
3   libcoreclr.dylib              	0x00000001068cd01e SEHExceptionThread(void*) + 142
4   libsystem_pthread.dylib       	0x00007fff67a2a109 _pthread_start + 148
5   libsystem_pthread.dylib       	0x00007fff67a25b8b thread_start + 15

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff6796d3d6 poll + 10
1   libcoreclr.dylib              	0x00000001068bfebe CorUnix::CPalSynchronizationManager::ThreadPrepareForShutdown() + 30
2   libcoreclr.dylib              	0x00000001068c1b50 CorUnix::CPalSynchronizationManager::WorkerThread(void*) + 912
[... 507 lines cut ...]
shared memory                      660K       20 
===========                     =======  ======= 
TOTAL                              3.6G     3982 
TOTAL, minus reserved VM space     3.2G     3982 

[... cut ...]
Bluetooth: Version 7.0.6f7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Composite Device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 55.3
Thunderbolt Bus: MacBook Pro, Apple Inc., 55.3

Hansaplast, 2021-07-28:
+1 

UVTools 2.15.1 (and previous version)
macOS Big Sur 11.4, Mac Pro 2013, 12 Core Intel Xeon, 64Gb RAM

sn4k3, 2021-07-29:
i will test on BigSur once i resume, it's stange catalina works and bigsur dont

PaleHaunt, 2021-07-29:
> i will test on BigSur once i resume, it's strange catalina works and bigsur dont

Catalina does NOT work, I'm running it on Catalina, and it crashes on open. ALSO, on 12.15 the "Exposure Time Finder" window flashes red opening, and then the window is not sizable, and you can't scroll down or across to get to the settings. You're essentially locked into a frozen window. I can start a new bug for this one.

sn4k3, 2021-07-29:
That's a known problem with mac reporting stupid scale factors and resolutions.
You need to uncheck this setting and save:

[image]

PaleHaunt, 2021-07-29:
> That's a known problem with mac reporting stupid scale factors and resolutions.
> You need to uncheck this setting and save:
> 

That did it for the window, thank you!

sn4k3, 2021-07-29:
> > That's a known problem with mac reporting stupid scale factors and resolutions.
> > You need to uncheck this setting and save:
> 
> That did it for the window, thank you!

BTW, can post a screenshot of your About window? Help -> About

PaleHaunt, 2021-07-29:
[image]

sn4k3, 2021-07-29:
Yeah like i said, 1792 x 1120 is a stupid resolution to report, not even your correct screen resolution... Ask apple why

sn4k3, 2021-08-02:
Fixed on latest release

closed 2021-08-02 by sn4k3