sn4k3/UVtools #245: .ctb file format

Britmetal, 2021-07-24:
Chitu has released a new version of Chitubox basic (1.9.0.), this new version requires printers that use the .ctb file format to upgrade to new versions of their firmware (4.4.3.), This new format is not compatible with UVTools.

Chitubox 1.9.0. Basic is here: https://www.chitubox.com/en/download/chitubox-free

Here is a link to where the new firmware is located https://www.chitubox.com/en/download/chitubox-free#firmwareList.

sn4k3, 2021-07-24:
CTBv4 is compatible with UVtools.
Only new boards with encryption are not (epax 5k and mega 8k)

Britmetal, 2021-07-26:
I beg to differ, When slicing with a .ctb file for Chitubox 1.9.0. UVTools cannot read it and sends the error attached

[image]

When slicing the same file with 1.8.1. there are not problems, after upgrading the printers firmware it will print both versions, unfortunately Github will not allow me to attach the .ctb files

sn4k3, 2021-07-26:
What printer you are using to slice?

Britmetal, 2021-07-26:
Phrozen Sonic Mini 4k

I have uploaded the specific files to my one drive with these links

https://1drv.ms/u/s!AnDZPDp_IB6giKVPHax06N4Aazgrwg?e=XSorjI
https://1drv.ms/u/s!AnDZPDp_IB6giKVQ0Q9VdPgR6Tm0Gg?e=GSNtR3

The Chitubox website for their Free version has details of what printers need firmware upgrades
https://www.chitubox.com/en/download/chitubox-free#firmwareList

sn4k3, 2021-07-26:
It works fine for me.
You just need to update UVtools for a more recent version, i always recommend the lastest

Britmetal, 2021-07-26:
Thanks, upgrading UVTools cured the problem, little bit confused, my old version 2.9.2, new version 2.1.5.

closed 2021-07-26 by Britmetal

sn4k3, 2021-07-26:
new version is **2.15.1**