sn4k3/UVtools #249: [FEATURE REQUEST] Support for Chitubox files used on Mars 3

jokeane, 2021-07-31:
Please support the file format used with the new version of Chitubox format used in the firmware on the new Mars 3 printer

sn4k3, 2021-07-31:
Mars 3 uses new encrypted file format. It wont happen so soon...

jokeane, 2021-08-03:
Chitubox / Elegoo said that they would allow free / open source software to write these files. Please contact them for info on how to do so.

sn4k3, 2021-08-03:
I've talked with chitu and i saw thier SDK announcement, still i'm unsure, the communication is not clear and everything seens cloudy, right now we need to wait to see what they will do.
My guess is that thier SDK would not be open-source, but in form of a closed dll dependency or a .exe which in that case does not comply with UVtools license nor the project. I will not incorporate private/closed dependencies under any term here as that broke the open-source and license of this project.
Chitu is going down the road, this encrypted thing is just stupid as useless, not to say that they are wasting CPU cycles and difficult decodes on motherboard where that precious cycles are needed for 8K. The manufacturers are hostages of chitu atm, another stupid thing... 
People should really shift from chitu products, they are downgrading the 3D experience and it future. I would never buy something that restrict me. For me i only buy gcode printers - Total freedom.

jokeane, 2021-08-03:
I agree with staying open source. I also agree encrypted files don't really serve a purpose in the standard workflow.There are ways to make open source projects use an encryption key without using a closed source binary. They probably aren't inventing the encryption system used, so you can probably just add a standard encryption library to the project.Ideally it is just a public key system and they just need to give you the specs and public key. If it is a private key, there are still ways to pull that key data in for the builds but not share it publicly in the repo.

sn4k3, 2021-08-03:
Yes, but the porpouse of that move from chitu is to prevent others from using thier format in a free way. If they give the specs how to read/write the file i would implement that directly of course. But my guess is that they want to cover everything and will not share the encryption scheme, otherwise the encryption will be just useless, as soon encryptions are known/reserved that encrytion become useless/dead/unsafe and only take aditional CPU cycles after all. It's is also possible to hack and reserve thier software, that happen in the past with ctbv3 encryption on layer data but that will help them because users will keep buying thier products, when what we need is a turnover.

jokeane, 2021-08-03:
If the reason for encryption is to prevent files from tampering (say if sliced files are shared online), then a public key encryption scheme can be 100% open and secure. Only the public key is exposed, which doesn't allow you to decrypt, only to encrypt new files.If the purpose is to protect market share by trying to lock in customers, then this strategy is doomed to fail. People will just install printer firmware which removes this restriction, or the printer companies will take such a hit to sales vs open systems they will ask to remove the encryption. Chitubox needs to compete on quality of features, not with some lame attempt to lock users into a walled garden.I plan to complain to Elegoo if they do not allow other software to create sliced files for the Mars 3. They are already getting pushback on YouTube reviews about the closed format.

winso2018, 2021-08-09:
we manufacture printers based on chituboard. A nightmare. Closed encrypted format, a new board is issued for each screen, format change without warning
We are developing our own motherboard, our own format that will be open. Unfortunately, we have few resources to complete the project quickly.

sn4k3, 2021-08-09:
> we manufacture printers based on chituboard. A nightmare. Closed encrypted format, a new board is issued for each screen, format change without warning
> We are developing our own motherboard, our own format that will be open. Unfortunately, we have few resources to complete the project quickly.

Good to see people shifting from this prison.
About format please see my DRAFT: https://github.com/sn4k3/UVtools/blob/master/Documentation/osla.md
I'm still composing but should be ready soon.

winso2018, 2021-08-09:
hi. we speak personally many time.
Now, we use self format (Zip, JSON based). In future, we can intrgrate open format. Now we not ready to do our project "fully open source"

sn4k3, 2021-08-09:
> Now, we use self format (Zip, JSON based). In future, we can integrate open format. 

Thats fine, theres also the UVJ (ZIP and Json), if going good CPU, zip is always best

> Now we not ready to do our project "fully open source"

The OSLA can be used on close situation, if you implement it, you only need to document the changes if any

wohali, 2021-08-24:
See: https://github.com/CryptoCTB/LiberatingMarsCLI

Looks like a clean way to support this.

sn4k3, 2021-08-24:
> See: https://github.com/CryptoCTB/LiberatingMarsCLI
> Looks like a clean way to support this.

UVtools support the file format some releases ago, it's just disabled and will continue to be

closed 2021-09-05 by sn4k3