sn4k3/UVtools #253: [FEATURE REQUEST] Formlabs .form File

shrimpyuk, 2021-08-01:
Hello!
I was wondering if Formlabs .form files could be added to UVTools? 
You can generate one by using their PreForm software.
I've attached an example of a .form file below.
[example.zip](https://github.com/sn4k3/UVtools/files/6913233/example.zip)

sn4k3, 2021-08-01:
That is from a SLA printer instead of LCD right?

shrimpyuk, 2021-08-01:
> That is from a SLA printer instead of LCD right?

Correct, they're SLA printers but are a little weird and the format is proprietary which sucks a little bit too

sn4k3, 2021-08-01:
Honestly i have no ideia how a SLA file should look or which properties they have.
I've inspected the file you sent, and file is encrypted and contain an CRC. 
The only hope here would be the Form guys sent a draft/documentation of the file format to include in UVtools, but i have a felling that won't happen... Including thier file format, means open doors to use other slicers, and they may not want that. However thier software looks good, do you get any problems with it? How SLA differ from LCD in problems, like elephant foot, islands, etc?

shrimpyuk, 2021-08-01:
If you take a look here they're the same https://formlabs.com/uk/blog/ultimate-guide-to-stereolithography-sla-3d-printing/ just different name for the same thing tbh.
I don't actually own a printer from them, however their software is AMAZING. The supports generated and the raft are perfect and theres a lot of pro's to it, however they really lock down their eco-system, you can only use their resin cartridges and their own software, kinda like an apple of the 3d printing world in a way.
Was hoping there was a way to extract the STL from a .form file, don't see why they lock it down tbh, cura is opensource and that gets a lot of publicity and gives me a great image of ultimaker, and if I ever needed a high end FDM machine that would set it in my eyes!

sn4k3, 2021-08-01:
SLA is not the same as LCD resin printer (DLP/mSLA)
Dots are cured by a laser dot per dot as beam pass by, while mSLA/DLP/LCD (Masked) cure all pixels at once using an image/LCD.
I guess that fileformat don't have any STL information there, slicer on PC read STL and generate a easy to read format for the printer, since read STL and supporting are a heavy job, in the end that file format should have like any machine code to move the laser to positions layer by layer. For example:

```
layer1:
X50 Y30
X50.035 Y30
X50.070 Y30
....
....
```

shrimpyuk, 2021-08-01:
Ohhh I see!!
Also from past builds, .form files used to have an embedded stl file as well as information about the print, but since they encrypt it to avoid the stl being extracted?

sn4k3, 2021-08-01:
The encryption in this case is more to avoid others to adopt or hack the file format, so you must strict use thier software. That work great when slicer is also great, so it give a "premium" felling, like apple.

shrimpyuk, 2021-08-01:
> The encryption in this case is more to avoid others to adopt or hack the file format, so you must strict use thier software. That work great when slicer is also great, so it give a "premium" felling, like apple.

Does this mean this is something you wouldn't add or something you can look into?
Their slicing and supports are seriously impressive, but sucks just how locked down it is

sn4k3, 2021-08-02:
> Does this mean this is something you wouldn't add or something you can look into?

Mean without thier cooperation this can't be added without serious effort.

X3msnake, 2021-08-02:
also sla requires a completely different type of slicer snd it is more akin
to fdm than MSLA

none of the tools in UVTools would support this ans basically we are
speaking of a completly new program dedicated to vector paths instead of
images.

X3msnake, 2021-08-02:
if u have a form1 i suggest you have a look at www.openmachines.xyz and the
preform community fork. if however you own any other form the hassle is
just too much and most people that own a formlabs most dont have the know
how or even care about fussing around with other softwares.

closed 2021-08-08 by sn4k3