sn4k3/UVtools #257: [BUG] Export to Creality CXDLP

gekiyakupl, 2021-08-05:
When trying to convert I get error (UVTools 2.16):

Convertion was not successful! Maybe not implemented...
System.Exception: Unable to detect the printer model from resolution, check if resolution is well defined on slicer for your printer model.
   at UVtools.Core.FileFormats.CXDLPFile.EncodeInternally(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\CXDLPFile.cs:line 582
   at UVtools.Core.FileFormats.FileFormat.Encode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 1747
   at UVtools.Core.FileFormats.FileFormat.Convert(Type to, String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 2445
   at UVtools.Core.FileFormats.FileFormat.Convert(FileFormat to, String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 2458
   at UVtools.WPF.MainWindow.<>c__DisplayClass93_0.<ConvertToOnTapped>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1359

sn4k3, 2021-08-05:
The error tell the problem.
Unable to detect the printer by resolution, meaning you have the wrong resolution set.
Post your printer settings print screen here

gekiyakupl, 2021-08-05:
Sorry about that, you are right I had set right size but wrong resolution in chitubox...

closed 2021-08-05 by gekiyakupl