sn4k3/UVtools #258: [BUG] Unix launch script has wrong line endings

Lenbok, 2021-08-08:
## System
 - OS: Ubuntu 20.04
 - Slicer: N/A
 - Printer: N/A

## Describe the bug

Quick bug report. I just downloaded the linux build UVtools_linux-x64_v2.16.0.zip and unpacked it. When I run the UVtools.sh, I get:

```
$ apps/UVtools_2.16.0/UVtools.sh
-bash: apps/UVtools_2.16.0/UVtools.sh: /bin/bash^M: bad interpreter: No such file or directory
```
The file is using windows line endings, and runs after fixing manually:
```
$ file apps/UVtools_2.16.0/UVtools.sh
apps/UVtools_2.16.0/UVtools.sh: Bourne-Again shell script, ASCII text executable, with CRLF line terminators
$ dos2unix apps/UVtools_2.16.0/UVtools.sh
dos2unix: converting file apps/UVtools_2.16.0/UVtools.sh to Unix format...
$ file apps/UVtools_2.16.0/UVtools.sh
apps/UVtools_2.16.0/UVtools.sh: Bourne-Again shell script, ASCII text executable
$ apps/UVtools_2.16.0/UVtools.sh
[launches OK]
```

sn4k3, 2021-08-08:
Nice cath, will fix on next release

sn4k3, 2021-08-08:
Fixed

closed 2021-08-08 by sn4k3