sn4k3/UVtools #261: [BUG] Exposure time finder LiftHeight Layer Logic is wrong

human890209, 2021-08-09:
## System
 - OS: Windows 10 x64
 - Slicer: Chitubox
 - Printer: QIDI 6.08 Mono

## Describe the bug
Example:
Bottom Layers Number: 4
Bottom/Normal LiftHeight:6mm
Current Output File:
Layer1 LiftHeight 6mm
Layer1 LiftHeight 6mm
Layer1 LiftHeight 6mm
Layer1 LiftHeight 6mm
Layer1 LiftHeight 6mm

## To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-08-09:
I dont see the problem, for me seens correct. Printer should lift whatever a diferent position is found for the first time with diferent image.

Layer3 (Bottom 4) Z+ LiftHeight 6mm  PositionZ=0.220mm 
Layer4 (Sample 1) Z+ LiftHeight 6mm  PositionZ=0.275mm [Different position + diferent image from above] -> Lift
Layer5 (Sample 2) Z LiftHeight 0.1mm  PositionZ=0.275mm [Same position + same image] -> No lift

human890209, 2021-08-09:
A different Z position is done by the previous layer's Lift and Retract, not the current layer's.

human890209, 2021-08-09:
I think this is not a BUG, please move it into Discussion.

sn4k3, 2021-08-09:
Bottom layers need other setting because they are often larger and more solid as they get more cure time and also to make sure first layer stick to build plate it need more care lift

> But the LiftOffDelay logic on Sample1 is not reasonable.

Let's keep it here until we figure out if is bug or not. 
NovaMaker for example uses lift after cure on thier gcode so layer0 will lift after cure. 
This require testing and can be different depending on shitty firmwares.
I will ask to someone test this out

human890209, 2021-08-09:
But the LiftOffDelay logic on Sample1 is not reasonable. LiftOffDelay is executed after the curing...

human890209, 2021-08-09:
> LightOffDelay is always performed after retract / before cure

WOW, everything cleared for me. Got my first Resin Printing Process Lesson today. Thanks a lot.

closed 2021-08-09 by human890209

sn4k3, 2021-08-09:
LightOffDelay is a mess and should never existed in first place... 
Chitu fixed that with ctb v4 with the rest time new fields

human890209, 2021-08-09:
It's should be called LiftOffDelay, the name is so misleading for the first layer. What light? Haha

sn4k3, 2021-08-09:
Not only you but almost everyone... Chitu really ruin this part. They should call it `LiftSequenceDelayTime ` because it what is, is a calculation of how much time motors take to complete the sequence + your desired extra time, now you see no one using this correctly without proper calculators. I dont really know why they introduce that mess

human890209, 2021-08-09:
I think the LightOffDelay's purpose is waiting for the resin to stop flooding. Retract Speed, resin's viscosity, build plate‘s size, and plate design(with holes or not) is the main factors. And the perfect result should be tested similar to the ExposureTime. But of course, the WaitTimeBeforeCure is much easier to understand and predict.

By the way, is the ctb v4 supported by UVTools?

sn4k3, 2021-08-09:
> I think the LightOffDelay's purpose is waiting for the resin to stop flooding. Retract Speed, resin's viscosity, build plate‘s size, and plate design(with holes or not) are the main factors. And the perfect result should be tested similar to the ExposureTime. But of course, the WaitTimeBeforeCure is much easier to understand and predict.

It is, but you must calculate the motor movement time to get the value right, read here: https://phrozen3d.com/apps/help-center/getting-started/what-does-light-off-delay-mean-when-i'm-setting-print-parameters

> 
> By the way, is the ctb v4 supported by UVTools? Many new fields there.

Yes, only TSMC is out of UVtools but that should not be used anyway.
Your printer if not included on list of chitubox v1.9.0 firmwares, can't use v4

human890209, 2021-08-09:
QIDI 6.08 Mono is supported by chitubox V4. But I think I will stick to the current firmware for now.
The only new feature I like is the delay after UV curing before Lift.
Light-off Delay's best value is from experiments. Increase or decrease 0.5s to make the resin not stable.  The calculation is just for guessing a start value.

sn4k3, 2021-08-09:
You should use it, because of the rest times. Older files will still print fine on new firmware.
Good values for common resins:
```
Wait before cure = 2.5s
Wait after cure = 1s
```

Wait after lift is useless/time waster

human890209, 2021-08-09:
Yes. I can't understand the purpose of a delay before retraction.

sn4k3, 2021-08-09:
> Yes. I can't understand the purpose of a delay before retraction.

Worst part is chitu release something and people blind use that stuff, i have seen people using a larger delay there than on before cure... 
TSMC is also useless the way is implemented

human890209, 2021-08-09:
The reason I stick to the current firmware is TSMC is not supported for now:>
TSMC actually let those Firmware Consts I mentioned about Lift&Retract in the QIDI 6.08 Mono discussion tweakable. Which indeed saves a lot of time.

sn4k3, 2021-08-09:
I don't think that's the case everywhere, saturn for example only have 1 lift sequence, no 2nd sequence on firmware. See this saturn video on ctbv3 using UVtools multiple exposures: https://www.facebook.com/groups/uvtools/posts/499511711158337/
As you can observe a lift of 0.1mm have no visible lift, if firmware does have a 2nd sequence you would see a additional lift there

human890209, 2021-08-09:
I'm pleased with my printer's 2mm LiftHeight_A and RetractHeight_B and 120mm/min LiftSpeed_B and 60mm/min RetractSpeed_B. I'm worried if I upgrade to ctb v4 I would lose it.

human890209, 2021-08-09:
I think I may create a script to add a virtual layer before each layer to get the Wait after cure delay!

human890209, 2021-08-09:
> In any case set it to 0 or use UVtools global WaitTimeBeforeCure property, it will set LightOffDelay accordingly
How to set the global property? Script?

sn4k3, 2021-08-09:
> I think I may create a script to add a virtual layer before each layer to get the Wait after cure delay!

Not the best thing, but may work

> How to set the global property? Script?

The `WaitTimeBeforeCure` ?

`SlicerFile.WaitTimeBeforeCure = 2.5; // Example`

human890209, 2021-08-10:
[image]
Here I found everything. TSMC and the best delays. Could these values be edited by UVTools' script?

sn4k3, 2021-08-10:
They can, but that will only work if you update your firmware with ctbV4

human890209, 2021-08-10:
Because my firmware got a TSMC-like motion, I will try those before heading towards the virtual layer method.

human890209, 2021-08-10:
How to change SlicerInfo in the script?
SlicerFile.BottomLiftDistance2 doesn't exist.

> uint EncryptionMode <fgcolor=cBlack, bgcolor=cRed>; // 0/8 for cbddlp files, 0xF (15) for ctb files, 0x2000000F (536870927) for v3 ctb and 1073741839 for v4 ctb files to allow per layer parameters

Does Encryption Mode the one that the FullEncode override to enable ctb file's per-layer settings?

sn4k3, 2021-08-10:
They will not work, those are only introduced on ctb v4. They are only here because UVtools follow the most recent ctb naming. For ctb v3 that would be padding fields (Useless fields)

human890209, 2021-08-10:
`
RestTimeAfterLift
RestTimeAfterLift2
`
Confusing name. Which one is the rest time after Curing before Lifting?

sn4k3, 2021-08-10:
yup, chitubox like confusing names xD

RestTimeAfterLift = Wait time after lift
RestTimeAfterRetract = Wait time before cure
RestTimeBeforeLift = Wait time after cure

human890209, 2021-08-10:
I got RestTimeAfterLift and RestTimeAfterLift2 in UVTools
[image]

sn4k3, 2021-08-10:
Yes, i dont know why, but they repeat rest time after lift on my findings, maybe a bug on chitubox

human890209, 2021-08-10:
I didn't find RestTimeBeforeLift. Is RestTimeAfterLift2 = RestTimeBeforeLift?

sn4k3, 2021-08-10:
> I didn't find RestTimeBeforeLift. Is RestTimeAfterLift2 = RestTimeBeforeLift?

No.

Again you need CTBv4 for that. Those are on new table: `PrintParametersV4`
`SlicerFile.PrintParametersV4` (Only present on ctb v4)

human890209, 2021-08-10:
I see. I notice I should not shift the LiftRetract into the next layer for my printer. I should shift the LightOffDelay into the previous layer :>

human890209, 2021-08-10:
I confirmed that my firmware support TSMC!
[image]

human890209, 2021-08-11:
I upgrade my firmware to ctb v4. A self-made fix script is still needed.

human890209, 2021-08-11:
The Rest Time system's per-layer settings are better than the virtual layer method. I upgrade to V4.
RestTimeAfterRetract(Wait Before Cure) is wrong if TSMC is used. The firmware starts timer after Retract2 but the slicer thinks the timer is started after Retract1.

Is there a way to not Cure a layer at all? I tried to set PWM down but it doesn't work. And Exposure time can't be set to 0s just like LiftHeight.