sn4k3/UVtools #262: [FEATURE REQUEST] Hope Pixel Dimming could work with Light Bleed Compensation

human890209, 2021-08-14:
Light Bleed Compensation is a reasonable tool but I choose Pixel Dimming to deal with Light Bleed for transparent resin and it works.

I need a thin 255 wall and some 255 infills to make the detail printable. I think the big solid gray areas are not reasonable for a fine-tuned ExposureTime with the Exposure time finder.
And Pixel Dimming can't be used after Light Bleed Compensation, cause it only dims pixels.

I wish Pixel Dimming could also add 255 walls and 255 infills to the layers with gray pixels. Maybe that should not be called Pixel Dimming : )
(Bug report: If choose Dim only walls for grayed layers, the infill will be lightened. That is not the result a user wants.)

sn4k3, 2021-08-14:
Both pixel dimming and Light Bleed Compensation works by subtracting pixels.
If you run Light Bleed Compensation and them pixel dimming, it will subtract from current pixel, making grey even more grey.

> (Bug report: If choose Dim only walls for grayed layers, the infill will be lightened. That is not the result a user wants.)

Confirmed, i will check that

human890209, 2021-08-14:
Thanks. Pixel Lightening is the requested feature : )

sn4k3, 2021-08-17:
Implemented

closed 2021-08-17 by sn4k3