sn4k3/UVtools #263: [BUG] Calibration ctb files can't open with Chitubox

human890209, 2021-08-15:
I tried to use Chitubox 1.9.0 to view the 3D result of ctb files from UVTools 2.18.0.
Most ctb files work. But the calibration files with a newly generated thumbnail image are invalid.
Maybe thumbnails are not the reason. Could be other Global Properties, such as EncryptionKey...

sn4k3, 2021-08-15:
Provide the problematic file

human890209, 2021-08-16:
I search "cbt" In this repository and got several results. Should they be replaced with "ctb"?

sn4k3, 2021-08-16:
> I search "cbt" In this repository and got several results. Should they be replaced with "ctb"?

Yes, CTB comes from **C**hi**T**u**B**ox

human890209, 2021-08-16:
After Exposure Time Finder or another calibration tool is generated, the thumbnail view window of UVTools is not updated automatically.
I could click the page left or page right button to update the thumbnail view manually, but I think it should update automatically if the thumbnail is replaced.

human890209, 2021-08-16:
And also a tool that could draw on the thumbnail image like the pixel editor (Draw, Text, Erase) is welcomed!

sn4k3, 2021-08-16:
> After Exposure Time Finder or another calibration tool is generated, the thumbnail view window of UVTools is not updated automatically.
> I could click the page left or page right button to update the thumbnail view manually, but I think it should update automatically if the thumbnail is replaced by the script.

I know a bout that. 
The Core can't access UI.
Not a big deal...

> And also a tool that could draw on the thumbnail image like the pixel editor (Draw, Text, Erase) is welcomed!
> Thumbnail Editor. Same UI (without Support and Hole) will do. Layer0 is Image1/2. Image2/2 could be generated from Image1/2.

That's a waste of time.
You can still upload and replace thumbnails by a image on  your disk

human890209, 2021-08-16:
How did the Properties and Layer Data UI update the values? By a fixed updated time?

sn4k3, 2021-08-16:
> How did the Properties and Layer Data UI update the values? By a fixed update interval?

Direct bound properties are via notify.
Layer data is updated every layer change
An other indirect bounds are via a forced full refresh of the UI

human890209, 2021-08-16:
I tried this code in my script:
`
            // Update the thumbnails
            Mat myThumbnail = SlicerFile.GetThumbnail().Clone();
            SlicerFile.SetThumbnails(myThumbnail);
`
The save-as ctb file can't open with Chitubox.

sn4k3, 2021-08-16:
Current version have a bug with v4. You need to wait for the patch

human890209, 2021-08-16:
`
Mat[] myThumbnail = new Mat[2];
myThumbnail[0] = SlicerFile.GetThumbnail(300).Clone();
myThumbnail[1] = SlicerFile.GetThumbnail(125).Clone();
SlicerFile.SetThumbnails(myThumbnail);
`
I tried these codes, the ctb file can be opened with Chitubox.
I guess CvInvoke.Resize(Thumbnails[i], Thumbnails[i], ThumbnailsOriginalSize[i]) corrupt something.

sn4k3, 2021-08-17:
Fixed

closed 2021-08-17 by sn4k3