sn4k3/UVtools #266: [BUG] Incorrect calculation MaterialMilliliters

Sat360, 2021-08-19:
## System
 - OS: [e.g. Windows 10 x64 v20H2]
 - Slicer: 
 - Printer: 

## Describe the bug
When I change layer height (by "Adjust layer height" or "Dynamic layer height") material milliliters in "Layer data" area become incorrect. And "Volume" in "I printed this file" window becomes incorrect too.

## To Reproduce
Steps to reproduce the behavior:
1. Go to 'Calibration - XYZ Accuracy'
2. Uncheck "Relief base with a center hole" and "Hollow model". Set the "X Length" and "Y Length" to 10mm and "Layer height" to 0.020mm
3. Generate test file.
4. Goto zero layer. Look at "Layer data". MaterialMilliliters is 0.0019ml. It is correct (10mm * 10mm * 0.02mm / 1000).
5. Goto "Tools - Adjust layer height".
6. Set "Modifier" to x5 (0.1mm). Click on "Adjust".
7. Goto Goto zero layer. Look at "Layer data". MaterialMilliliters is 0.0019ml. It is incorrect. Right value is 0.010 (10mm * 10mm * 0.100mm / 1000).

## Expected behavior
MaterialMilliliters should change when I change layer height.

## Screenshots
[image]

sn4k3, 2021-08-19:
I confirm the bug, per layer value isn't calculating, it only calculate when image is changed. I will fix that.

sn4k3, 2021-08-21:
fixed

closed 2021-08-21 by sn4k3