sn4k3/UVtools #268: [BUG] Rerf test goes on a loop

Liger0, 2021-08-19:
## System
 - OS: Windows 10 x64
 - Slicer: 
 - Printer: Anycubic photon mono se

## Describe the bug
If you try to open anycubic's rerf test file, the software goes on a loop, and it keeps restarting the "computing issue 8resin traps)" every time it reaches 51%.

## To Reproduce
Steps to reproduce the behavior:
1. Open file
2. Select rerf test
4. Loop

## Expected behavior

## Screenshots

## Files
[R_E_R_F.zip](https://github.com/sn4k3/UVtools/files/7014093/R_E_R_F.zip)
 
## Exceptions Log file

## Additional context

sn4k3, 2021-08-19:
Nothing wrong here. It just take a ton of time to process due the amount of holes on the model.
Resin trap don't know before hand how many traps exists, so it start on first hole, and start to navigate up and down, raising the progress (Processed layers) but those processed layers are for only 1 hole! When it find another hole on other path it will start over again from that layer and go up/down, repeating up to deal with all holes. Thats why progress goes up and down. So progress is per hole and not a whole.

It's not stuck in loop, just a ton of holes to process... On my pc it took a good amount of minutes to finish

closed 2021-08-21 by sn4k3

tslater2006, 2021-09-16:
Hey @Liger0, you should give this a shot in the latest version :) 2.22.0 has a brand new resin trap detection, motivated directly by this issue, and in fact that R_E_R_F.zip file was one of our main tests :)