sn4k3/UVtools #270: [BUG] Memory error using Sonic4K

MidniteM, 2021-08-22:
## System
 - Slicer: Prusa
 - Printer: Sonic 4k
 -

## Describe the bug
Latest version 2.19.2 causing memory fail error when trying to print.

## To Reproduce
Have sliced with Prusa then file conversion to ctb v4. 
Check and repair layer issues
Edit layer parameters
Apply pixel dimming
Save file to USB

On printer S4K:
Print file
Appears fine, picture displays, exposure settings show correct.
Before first layer prints, memory error occurs and plate moves back up.

Tried with slice in chitubox with TSMC settings already loaded
Fixed layers, applied pixel dimming
Save file.
Same printer behavior

Attempted a different USB, same error.

Previous version of UVtools before TSMC implemented was working.

Thankyou for this awesome tool

sn4k3, 2021-08-22:
Maybe pixel dimming is your problem.
Can you try same file but without apply pixel dimming?
CTB does not goes well with checker pattern, no memory for that...

MidniteM, 2021-08-22:
Thanks for the fast reply, I'll give it a try.

Pixel dimming is the only way I can get quality prints from my S4K so I
edit all my files to brightness 190.

Also - the pixel dimming function in latest version seems backwards? Before
I set brightness to 190 (about 75%) - but now I set it to 65 (255-65 =190)
for same result - is this the correct way to do it?

Cheers!

sn4k3, 2021-08-22:
Ah you are not using a pattern then? Only a solid?
New pixel arithmetic uses a different logic yes. You can choose Set, subtract or add a brightness.
If you are on set mode, every pixel will set to your desired value.
On subtract will subtract pixel with your value, and same for add.

Old pixel dimming was inverted, it was subtracting too, but 255 was inverted to 0 and 0 to 255.

MidniteM, 2021-08-22:
Unfortunately that didn't work.
This is the slice converted from Prusa but no pixel dimming applied.

Thankyou for your time, very much appreciated.

MidniteM, 2021-08-22:
That's right, only solid.

Yes she the subtract is just inverted, so my new value of 65 is the same as
the old 190?

sn4k3, 2021-08-22:
Yes: 255-190 = 65 in subtract mode.

Also try print the file with ctb v3

MidniteM, 2021-08-22:
Sliced a test file in Prusa
Loaded in UVtools and converted to ctb v3
Saved to USB
S4K starts the print and appears to be working fine (exposure values etc
all appearing in the settings)
Layer 1 tried to print - UV came on, but black LCD and no image displayed,
only for normal layer exposure time (2 seconds), and then raised to normal
height and paused for over a minute at which point I cancelled it.

sn4k3, 2021-08-22:
Ok provide me the file.

> Previous version of UVtools before TSMC implemented was working.

Which one?

MidniteM, 2021-08-22:
I just recreated the same test using UVtools version 2.18.0 and the file appears to be printing fine.
Attached both files here - email didnt seem to work.
[Testfiles.zip.zip](https://github.com/sn4k3/UVtools/files/7026468/Testfiles.zip.zip)

sn4k3, 2021-08-22:
Which one is the working and the bad?

MidniteM, 2021-08-22:
The one marked with a 2 is the working copy from 2.18.
Thankyou

sn4k3, 2021-08-22:
Fixed, please try

closed 2021-08-22 by sn4k3

MidniteM, 2021-08-22:
Tested with my workflow - slice in Prusa, convert to ctb v4, check/dim/edit/save - print.
Works perfectly! 
Absolute legend! thankyou :)