sn4k3/UVtools #278: [BUG] All layers same exposure time (bottom exposure time) for .pwmo (Anycubic Moni)

th-m-vogel, 2021-08-29:
## System
 - OS: Windows 10 x64
 - UVtools Version: 2.20.3 X64
 - Slicer: Photon Workshop 
 - Printer: Anycubic Photon Mono

## Describe the bug
Any slices are exposed the same time, what is in this case the bottom layer exposure time.

## To Reproduce
Steps to reproduce the behavior:
1. Slice a object (Photon Workshop or Chitubox)
2. import in UVtools
3. Save
4. print the file saved from UVtools, everything is hard overexposed
5. extract file contes with UVtools
6. check layers.ini

## Expected behavior
non bottom layer are exposed by standard exposure time

## sample data from configuration.ini
[Header]
Section = [Mark: HEADER, Length: 80]
PixelSizeUm = 51
LayerHeight = 0.05
LayerExposureTime = 3.2
LightOffDelay = 0.4
BottomExposureSeconds = 20
BottomLayersCount = 4
LiftHeight = 8
LiftSpeed = 3
RetractSpeed = 6
VolumeMl = 3905.846
AntiAliasing = 1
ResolutionX = 1620
ResolutionY = 2560
WeightG = 0
Price = 0.25521803
PriceCurrencyDec = 36
PriceCurrencySymbol = $
PerLayerOverride = 0
PrintTime = 37980
Padding1 = 0
Padding2 = 0

## sample data from layers.ini, layer 10 what is not a bulttom layer
[10]
NonZeroPixelCount: 233335
BoundingRectangle: {X=528,Y=626,Width=564,Height=985}
IsBottomLayer: False
LayerHeight: 0.05
PositionZ: 0.55
ExposureTime: 20
LiftHeight: 8
LiftSpeed: 180
MaterialMilliliters: 0.0303ml (0.00%)

sn4k3, 2021-08-29:
I tried to replicate this, but it works for me.
Can you send me your file both before and after save with uvtools?
Here my output:

```
[FileMark]
Mark = ANYCUBIC
Version = 1
AreaNum = 4
HeaderAddress = 48
Padding1 = 0
PreviewAddress = 144
PreviewEndAddress = 75436
LayerDefinitionAddress = 75436
Padding2 = 0
LayerImageAddress = 84416

[Header]
Section = [Mark: HEADER, Length: 80]
PixelSizeUm = 50
LayerHeight = 0.025
LayerExposureTime = 1
LightOffDelay = 12.53
BottomExposureSeconds = 10
BottomLayersCount = 3
LiftHeight = 7
LiftSpeed = 1.33
RetractSpeed = 1.33
VolumeMl = 752.518
AntiAliasing = 1
ResolutionX = 3840
ResolutionY = 2400
WeightG = 0
Price = 0
PriceCurrencyDec = 36
PriceCurrencySymbol = $
PerLayerOverride = 1
PrintTime = 7623
Padding1 = 0
Padding2 = 0

[0]
NonZeroPixelCount: 354596
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: True
LayerHeight: 0.025
PositionZ: 0.025
ExposureTime: 10
LiftHeight: 15
LiftSpeed: 19.8
MaterialMilliliters: 0.0222ml (0.00%)

[1]
NonZeroPixelCount: 354806
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: True
LayerHeight: 0.025
PositionZ: 0.05
ExposureTime: 10
LiftHeight: 15
LiftSpeed: 19.8
MaterialMilliliters: 0.0222ml (0.00%)

[2]
NonZeroPixelCount: 354633
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: True
LayerHeight: 0.025
PositionZ: 0.075
ExposureTime: 10
LiftHeight: 15
LiftSpeed: 19.8
MaterialMilliliters: 0.0443ml (0.01%)

[3]
NonZeroPixelCount: 354657
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: False
LayerHeight: 0.025
PositionZ: 0.1
ExposureTime: 1
LiftHeight: 7
LiftSpeed: 79.8
MaterialMilliliters: 0.0665ml (0.01%)

[4]
NonZeroPixelCount: 354835
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: False
LayerHeight: 0.025
PositionZ: 0.125
ExposureTime: 1
LiftHeight: 7
LiftSpeed: 79.8
MaterialMilliliters: 0.0222ml (0.00%)

[5]
NonZeroPixelCount: 354996
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: False
LayerHeight: 0.025
PositionZ: 0.15
ExposureTime: 1
LiftHeight: 7
LiftSpeed: 79.8
MaterialMilliliters: 0.1109ml (0.01%)

[6]
NonZeroPixelCount: 355153
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: False
LayerHeight: 0.025
PositionZ: 0.175
ExposureTime: 1
LiftHeight: 7
LiftSpeed: 79.8
MaterialMilliliters: 0.1332ml (0.02%)

[7]
NonZeroPixelCount: 355306
BoundingRectangle: {X=1393,Y=749,Width=1054,Height=902}
IsBottomLayer: False
LayerHeight: 0.025
PositionZ: 0.2
ExposureTime: 1
LiftHeight: 7
LiftSpeed: 79.8
MaterialMilliliters: 0.1554ml (0.02%)
```

th-m-vogel, 2021-08-29:
> Hi,
>
> thanks for fast Response. I realized something is wrong when printing some of the included calibration tests, i.e. Elefant foot calibration.
>
> Slicing with Photon Workshop and direct going to print, I never had a Problem.
> Attached a pwmo sliced with Photon Workshop and the same after Opening and saving as with UVtools.
>
> Also, the extracted ini data from UVtools.
>
> I also attached a ElefanfFoot test generated in UVtools same issue, all layers at 20 seconds
>
> One Question, how to extract the data from a pwmo file without the use of UVtools?
>
> Best Regards, Thomas
>
> Screenshot UVtools, elefant foot, layer 34, IsBottomLayer: False, ExposureTime: 20s

Original mail, Message size exceeded error … following up with multiple mail, single file attached.

th-m-vogel, 2021-08-29:
Hi,

additional info, UVtools opens with the UVtools_demo_file. Converting this to Photon Mono (pwmo) does not produce this bug!

sn4k3, 2021-08-29:
I found the bug, it's general, it happen on all files on calibration tests

th-m-vogel, 2021-08-29:
Hi,

Update, It happens only to pwmo files sliced using Photon Workshop. Scliced using chitubox or Lychee it does not happen. Attached a Photon Workshop slice

sn4k3, 2021-08-29:
Fixed on new release

closed 2021-08-29 by sn4k3