sn4k3/UVtools #279: [FEATURE REQUEST] Ability to limit CPU usage; # of cores, priority, etc

apullin, 2021-08-30:
When I use UVTools to detect problems in a 4K output for a PWMX format file, it appears to use 100% of all CPU time.
Having some option to limit the number of cores or threads that will spawn would be great.

This might also be represented by some backlog item to shift to more async execution to alleviate the same high CPU time issue.

sn4k3, 2021-08-30:
That's the way a good program works, take all your CPU for workload and invest on the problem, wasting a CPU is wasting time. I'm afraid you have a weak CPU or laptop? This kind of program require good machines and when you start entering in 4K and 8K theres a big slow down on processing stuff.

apullin, 2021-08-30:
Interesting. I disagree with that assessment. Computationally heavy tasks should coexist with other resident programs. Although the OS is responsible for enforcing this, too.
Note that the CPU is not "wasted": it is available for other needs of the system. This is, after all, a desktop application.
This is a standard feature in tools like Xilinx Vivado, Blender rendering, etc.

And, of course, I do not know how much of the behavior is driven by the underlying library calls themselves, e.g. OpenCV.

For throughput or latency, sure, the computation should be the highest priority.
But are throughput or latency the primary factors or the only factors here? I would argue that usability is relevant.

Only my Core i9 machine, 12-thread capable, the compute steps go to 1200% usage, to the point of even lagging the UI itself.

Also of note: I am running on OSX, so the behavior I am describing may be worse than on other platforms (due to the whole dotNet framework port).

sn4k3, 2021-08-31:
Your Mac is laptop or desktop? X64 or Arm?
In fact i always had a setting to limit the number of threads in Parallel functions but i never applied that, atm is up to system manage, it will spawn how many he thinks its best, for example it can spawn only 4 if others are busy. But I can implement that and opt for user choose and limit the number of cores at use in parallel operations to relief the load. Other than that is all up to OS to manage.
On my i9-9900k on heavy computations it also delay other programs but is usable. 
Also keep in mind if you limit to half of cores for example, programs you may use if OS delegate the same cores, it will slow down programs on the same cores, here is where you hope OS or framework delegate more free cores to the task and not the others being used.

Also one problem that hits the performance very bad is the heat, you need to keep the CPU cool or it will threshold and lose a lot of power, we all know mac is not on pair with a good pc desktop with decent custom heatsinks/AIOs, and if is a laptop forget, they dont have enough power nor heat dissipation to compete against a desktop on that field. So i would run a hardware monitor to see how far temperature goes on that situations.

Btw if you can post a screenshot of the benchmark utility of the UVtools with "4K Random Encode CTB".

Here a example: 
[image]

Even at full load, some cores never hit max usage, and current usage still have some room. This can only happen with suficient cooling, also note that i have my CPU overclocked to 5GHz

sn4k3, 2021-08-31:
Will configure as follow:

[image]

apullin, 2021-08-31:
CPU is x86 in a Macbook Pro laptop.
Of note is that the CPU does appear to be oversized for the thermal capacity of this computer. Thermal throttling happens all the time.

Here is the result of running the benchmark:
[image]

And the usage reported by htop:
[image]

Notably, it does not hit the CPU as hard as it can, for all the cycles on all the cores.

However, when I do load a big 4K print, during the problem detection, it does use up every cycle:
[image]

As mentioned, it is also entirely possible that the dotNet framework port for OSX is not handling async thread dispatch of task/threads well, so it results in an over-use of the CPU at times.

The proposed change looks great, and I'll give it a try as soon as it is rolled out.

sn4k3, 2021-08-31:
> Of note is that the CPU does appear to be oversized for the thermal capacity of this computer. Thermal throttling happens all the time.

Yeah the performance seens good, for sure they are boosting it, but on a laptop we can't expect a thermal miracle, with poor cooling it will throttling on heavy load, most likely lag comes from there, heavy load under a caped cpu always produce that effect.

> Notably, it does not hit the CPU as hard as it can, for all the cycles on all the cores.
> However, when I do load a big 4K print, during the problem detection, it does use up every cycle:

Yes, the benchmark is a loop on same operation, at some point some most of the instructions are cached.
The problem detection however is a different beast...

> As mentioned, it is also entirely possible that the dotNet framework port for OSX is not handling async thread dispatch of task/threads well, so it results in an over-use of the CPU at times.

Not impossible but i doubt, the algo is really heavy, 4K decodes/encodes x number of layers x passes, for sure every laptop have this same "problem" no matter if windows or linux or macos. In fact as i told you, on my desktop it does start to lag on heavy files, it's usable for normal exploring, but for example, if i have a movie playing, it will lag, skip, sound jump, etc. So for sure it just taking all your CPU and apply it to the problem, i never tried to put UVtools as low-priority on task manager and see how it goes, but on all my attempts with other problems seens to have no effect... Also UVtools uses GPU too, thanks to that image decode/encodes are delegated to GPU, so it also matters, if you have a dedicated GPU that workload is relief from CPU which is a bless.

I don't see many programs giving you the option to limit the cores it can use, for example, HandBreak causes same effect as UVtools making PC much more slower, and video reencoding takes forever.

PS: NET 6.0 will give a huge boost on performance, and mac will also benefit a lot from it. Waiting for stable release to go out :)

> The proposed change looks great, and I'll give it a try as soon as it is rolled out.

Will release in some minutes so you can try, i did, i set to 14 seens ok to keep PC working without lag, didnt try with 15 but hyperthreading is not the same as a physical core thats why i set to core count - 2.

UVtools is one of few programs that system specs are never in overkill, the more you have the more it will use 😁 I already feel i need a 64 core system xD
And wait for painfull 8K images...

sn4k3, 2021-08-31:
Here my results if i goes from 16 to 1 thread on a 2K file (Computing issues):

[image]

```
#	Started	Time(s)	Description
23	16:42:13	44.05	Computing Issues
22	16:41:41	24.01	Computing Issues
21	16:41:20	17.34	Computing Issues
20	16:41:01	14.3	Computing Issues
19	16:40:44	12.29	Computing Issues
18	16:40:28	11.08	Computing Issues
17	16:40:11	10.08	Computing Issues
16	16:39:58	9.61	Computing Issues
15	16:39:45	8.93	Computing Issues
14	16:39:31	8.39	Computing Issues
13	16:39:16	8.26	Computing Issues
12	16:39:05	7.95	Computing Issues
11	16:38:50	7.74	Computing Issues
10	16:38:35	7.45	Computing Issues
9	16:38:11	7.54	Computing Issues
8	16:37:50	7.61	Computing Issues

```

CPU usage drop as i thread count decreases, fans stop to kick in on the 4 core and below.

apullin, 2021-08-31:
OK, the change works great.

Here is `htop` with 12/16 threads as the setting:
[image]

The OS doesn't lag, background processes get a few cycles and don't hiccup, progress bar runs smoothly.

sn4k3, 2021-08-31:
Yeah i think we can close this issue.
It does the difference, you can also try the "!" preset. I guess it will do just fine as well, but would not give huge benefit either

closed 2021-08-31 by sn4k3