sn4k3/UVtools #280: [FEATURE REQUEST] Adjust Layer Height on specific layers

Maeyanie, 2021-09-03:
It would be useful if the "Adjust Layer Height" dialogue had the option to select only specific layers, in the same way most of your other dialogues do.

This would be useful to create a thicker or thinner first layer to help with plate adhesion, or to have a "fast range" with thicker layers to speed up the early stage of a print which contains only supports.

Alternately, it would also work to be able to set the layer height manually in the "Edit Print Parameters" dialogue with "Change settings per a layer range" checked, but the former idea sounds like it would be less work to implement.

sn4k3, 2021-09-03:
The "Adjust Layer Height" is to change layer height globaly in a safe way that work with all printers and then is restricted to multiples of valid height.
Also keep in mind if you change layer height you must keep model integrity.

To do what you want, slice a file at 0.01mm layers and use the "Dynamic layer height" tool, it will thicken and optimize where it can in a auto way. 
But notice that you require different exposures per different thicknesses, for example, 0.05 layer exposure on a 0.10 layer wont do in my printer case (1.8s vs 3s required for 0.10) so you must have a table of usable times per thickness or it will not give the optimal result

Maeyanie, 2021-09-04:
Okay, it's a bit clunky to force a manual layer height with that, but it looks possible. I'll close this.

closed 2021-09-04 by Maeyanie

sn4k3, 2021-09-04:
Force layer height will rarely work as you need multiples to ensure the total height remains the same.
You dont need to thinker with dynamic layer height, just slice at 0.01mm and let uvtools to optimize and group similar layers, you can always set a min/max layer height

Maeyanie, 2021-09-04:
That just sounds like a recipe for bad print lines to me.

I was able to get it to do thicker raft and support layers by setting the min and max to both equal the desired layer thickness (2x the regular layer thickness in this case), so it will work. Thanks.

sn4k3, 2021-09-04:
Give it a try one day ;)