sn4k3/UVtools #285: [BUG] Resin trap issue detection

jimokeane, 2021-09-07:
Latest version seems to get stuck computing issues on the resin trap detection step. It just bounces up / down over the same layers and never finishes.

## System
 - OS: MacOS 10.15.7
 - Slicer: Lychee
 - Printer: Anycubic Photon (.photon file)

## Describe the bug
Latest version seems to get stuck computing issues on the resin trap detection step. It just bounces up / down over the same layers and never finishes.

## To Reproduce
Steps to reproduce the behavior:
1. Load Lychee sliced .photon file

## Expected behavior
Not going into infinite loop

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-09-07:
Most probable is your file is complex with to much holes.

See:

1. https://github.com/sn4k3/UVtools/issues/268
2. https://github.com/sn4k3/UVtools/issues/221

Attach your file if you want futher investigation

sn4k3, 2021-09-16:
New release give more performance.

closed 2021-09-16 by sn4k3