sn4k3/UVtools #286: [BUG] Error converting any file to Chitubox CTBv4 on MacOS

madeinoz67, 2021-09-09:
## System
 - OS: macos 11.5.2
 - Slicer: Lychee
 - Printer: Photon S
 - UVTools: v2.21.1

## Describe the bug
When trying to convert any file to ctbv4 I receive an error "The save file operation failed" (screenshot attached)

When converting to other formats all is as expected

## To Reproduce
Steps to reproduce the behavior:
1. Click on File menu 
2. Click on convert to...
3. Scroll down to '.convert to Chitubox V4 format
4. delay is seen then error dialog appears

## Expected behavior
file dialog should open and save to new format.

## Screenshots

[image]

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-09-09:
That error is not from UVtools. Something on mac is not allowing to do the operation.
Try to convert to .rgb.cws does same message shows?

madeinoz67, 2021-09-14:
thanks for that, yes same error as above for "Creation Workshop X" .xml.cws  so will track things down locally.

sn4k3, 2021-09-14:
ok so mac is triping when i have two extensions on filters. But i tested on my VM and it works. BigSur X64.

sn4k3, 2021-12-26:
Fixed

closed 2021-12-26 by sn4k3