sn4k3/UVtools #299: [BUG] Settings not being saved (Mac)

rfield19, 2021-09-19:
## System
 - OS: Mac OSX 11.xx

## Describe the bug
I was experimenting with the new network sending features - which work great btw! - but the settings for the network entries and volume shares are not being saved.  The next time that UVtools is run, the network entries and volume shares sections are completely blank again.

edit: It may actually be that uvtools isn't saving *any* setting changes at all ... 
I've tried changing a few other settings like 'load demo file if no file loaded' and it doesn't remember that setting after quitting either. (I'm pressing 'save' button after adding/changing an entry of course!)

## Expected behavior
network shares and networked printer settings (and other settings) to be saved.

sn4k3, 2021-09-19:
Open UVtools, Menu -> Help -> Open settings folder

- Does folder open?
- Is blank?
- contain usersettings.xml file?

rfield19, 2021-09-19:
Hi,
when I choose the 'open settings' option from the menu, nothing (at all) happens.
No finder window opens, etc...

>> Bene nova?
Sorry, meant I'm just using the 'send to custom locations' option with an smb share into a mariner usb drive.
I've not tried the direct to printer network sending! (printer is an epax e10)

sn4k3, 2021-09-19:
Ok ARM can be the cause since UVtools is not yet compatible with ARM, it run in emulation mode.
If folder doesn't open it mean it doesn't exists.
On a x64 settings are save here:
[image]

You can try to create a manual UVtools folder on that path and paste a usersettings.xml file and see if it loads, but probably not..
[usersettings.zip](https://github.com/sn4k3/UVtools/files/7192445/usersettings.zip)
.

rfield19, 2021-09-19:
Curiously there was already a UVtools directory in the ~/ location - dated mid July - so it perhaps had access to that directory in an earlier version.  There was an old errors.log file in there.

Anyway, 
i) I checked the permissions, set them to 777 - no difference.

ii) I copied in the usersettings xml from your archive - again, no difference.  

iii) I manually altered the xml to read
<LoadDemoFileOnStartup>false</LoadDemoFileOnStartup>
and it still made no difference on startup!

So I guess it's unfortunately not accessing that directory/usersettings.xml file at all.
I had a look in OSX console and turned on the realtime diags, but it didn't generate any message when I clicked the 'save' button in uvtools.

sn4k3, 2021-09-19:
WIthout a ARM test machine will be very hard to me to debug this. I use absolute path for files, so that should be ok. 
If you can save files i see no reason why not it wont save settings, nor recent open files i guess?

rfield19, 2021-09-19:
Yes, that's right - the File -> Open recent option in the menu displays the same behaviour - does nothing at all when you click on it.  No issue with saving out files after repair or conversion though.

I understand it being difficult to test ARM code ; but if you have any brain waves later on and want me to try anything out, just let me know!

sn4k3, 2021-09-19:
The only way is sending you custom builds to test out. But is a pain to build a test lines of codes in that way

sn4k3, 2021-09-19:
Try this build: https://we.tl/t-jVxgMLaYlS

Open with terminal and see if any message is told after you save Settings. If not look at desktop for a txt.

rfield19, 2021-09-19:
Thanks, that was quick :)

Some progress...
I ran it from the ~/Downloads directory to start with - and although it didn't output anything to the terminal, it _did_ create a UVtools directory in ~/Downloads (not in ~/), and did also correctly output a usersettings.xml file when I clicked the save button.

After some experimentation though, it creates the UVtools config directory in whatever location I was when running the command from at the time.

For example if I was in ~/temp directory in terminal and then ran /Applications/UVtools.app/Contents/MacOS/UVtools then it would create the UVtools config directory in ~/temp.

So rosetta is not liking the absolute paths in the code?

sn4k3, 2021-09-19:
Humm... maybe it's returning empty path of special folders, try this and observe the terminal, it should output the folder where it should save things... https://we.tl/t-0d4ys3qNkq

rfield19, 2021-09-19:
Yes, seems to be missing off the path prefix :/

If I run 
./UVtools.app/Contents/MacOS/UVtools
from the Downloads directory I now get an output of

_UVtools/usersettings.xml_

(and same thing in a uvtools.txt file on the desktop)

sn4k3, 2021-09-19:
Ok so its a problem in .NET, not really a problem in fact, because .NET5.0 doesnt support Rosetta. This will be proper fixed with .NET6.0.
I'm using the special folder of: `Environment.SpecialFolder.LocalApplicationData` which seens missing on Rosetta.
The only fix i can find, is use other folder that .NET is aware and switch to that if got a empty folder.

Please run this and paste here the output: https://we.tl/t-9YNEKOtXI7

rfield19, 2021-09-19:
OK, here you go.  I now get this as output:

Desktop: /Users/rupert/Desktop
Programs: 
MyDocuments: /Users/rupert
MyDocuments: /Users/rupert
Favorites: /Users/rupert/Library/Favorites
Startup: 
Recent: 
SendTo: 
StartMenu: 
MyMusic: /Users/rupert/Music
MyVideos: 
DesktopDirectory: /Users/rupert/Desktop
MyComputer: 
NetworkShortcuts: 
Fonts: /Users/rupert/Library/Fonts
Templates: 
CommonStartMenu: 
CommonPrograms: 
CommonStartup: 
CommonDesktopDirectory: 
ApplicationData: /Users/rupert/.config
PrinterShortcuts: 
**LocalApplicationData:** 
InternetCache: /Users/rupert/Library/Caches
Cookies: 
History: 
CommonApplicationData: /usr/share
Windows: 
System: /System
ProgramFiles: /Applications
MyPictures: /Users/rupert/Pictures
UserProfile: /Users/rupert
SystemX86: 
ProgramFilesX86: 
CommonProgramFiles: 
CommonProgramFilesX86: 
CommonTemplates: 
CommonDocuments: 
CommonAdminTools: 
AdminTools: 
CommonMusic: 
CommonPictures: 
CommonVideos: 
Resources: 
LocalizedResources: 
CommonOemLinks: 
CDBurning:

sn4k3, 2021-09-19:
Ok a ton of missing directories that .NET is aware.
Best candidate looks like: ApplicationData: /Users/rupert/.config

Can you confirm if that folder exists on system, and the contents? eg screenshot of folder contents

rfield19, 2021-09-19:
It exists, yes - and appears to be writable as a regular user too.

_rupert@Mac-mini ~ % ls -R .config_
Autodesk	iterm2

.config/Autodesk:
FusionFabrication.FusionNesting

.config/iterm2:
AppSupport

_rupert@Mac-mini ~ % cd .config
rupert@Mac-mini ~ % touch test_

_rupert@Mac-mini .config % ls -la_
total 0
drwxr-xr-x   3 rupert  staff   96 10 Aug 12:36 Autodesk
drwxr-xr-x   3 rupert  staff   96  7 Jul 00:22 iterm2
-rw-r--r--   1 rupert  staff    0 19 Sep 21:40 test

sn4k3, 2021-09-19:
This build should fix that: https://we.tl/t-vSfBJiD8tw

i'm bullet proof this with following:

```
var folder = Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData);
if (string.IsNullOrWhiteSpace(folder)) folder = Environment.GetFolderPath(Environment.SpecialFolder.ApplicationData);
if (string.IsNullOrWhiteSpace(folder)) folder = Environment.GetFolderPath(Environment.SpecialFolder.CommonApplicationData);
if (string.IsNullOrWhiteSpace(folder)) folder = Environment.GetFolderPath(Environment.SpecialFolder.UserProfile);
var path = Path.Combine(folder, About.Software);
```

For reference here the windows 10 path:

```
Desktop: D:\Tiago\Desktop
Programs: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\Start Menu\Programs
MyDocuments: D:\Tiago\Documents
MyDocuments: D:\Tiago\Documents
Favorites: C:\Users\tiago\Favorites
Startup: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Startup
Recent: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\Recent
SendTo: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\SendTo
StartMenu: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\Start Menu
MyMusic: D:\Tiago\Music
MyVideos: D:\Tiago\Videos
DesktopDirectory: D:\Tiago\Desktop
MyComputer: 
NetworkShortcuts: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\Network Shortcuts
Fonts: C:\WINDOWS\Fonts
Templates: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\Templates
CommonStartMenu: C:\ProgramData\Microsoft\Windows\Start Menu
CommonPrograms: C:\ProgramData\Microsoft\Windows\Start Menu\Programs
CommonStartup: C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Startup
CommonDesktopDirectory: C:\Users\Public\Desktop
ApplicationData: C:\Users\tiago\AppData\Roaming
PrinterShortcuts: 
LocalApplicationData: C:\Users\tiago\AppData\Local
InternetCache: C:\Users\tiago\AppData\Local\Microsoft\Windows\INetCache
Cookies: C:\Users\tiago\AppData\Local\Microsoft\Windows\INetCookies
History: C:\Users\tiago\AppData\Local\Microsoft\Windows\History
CommonApplicationData: C:\ProgramData
Windows: C:\WINDOWS
System: C:\WINDOWS\system32
ProgramFiles: C:\Program Files
MyPictures: D:\Tiago\Pictures
UserProfile: C:\Users\tiago
SystemX86: C:\WINDOWS\SysWOW64
ProgramFilesX86: C:\Program Files (x86)
CommonProgramFiles: C:\Program Files\Common Files
CommonProgramFilesX86: C:\Program Files (x86)\Common Files
CommonTemplates: C:\ProgramData\Microsoft\Windows\Templates
CommonDocuments: C:\Users\Public\Documents
CommonAdminTools: C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Administrative Tools
AdminTools: C:\Users\tiago\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Administrative Tools
CommonMusic: C:\Users\Public\Music
CommonPictures: C:\Users\Public\Pictures
CommonVideos: C:\Users\Public\Videos
Resources: C:\WINDOWS\resources
LocalizedResources: 
CommonOemLinks: 
CDBurning: C:\Users\tiago\AppData\Local\Microsoft\Windows\Burn\Burn
```

rfield19, 2021-09-19:
Thanks Tiago, you've fixed it - all works great now.  Amazing as ever :)

sn4k3, 2021-09-19:
> Thanks Tiago, you've fixed it - all works great now. Amazing as ever :)

Great, will be on next public release. 
Have fun, that build already have some new stuff that you might notice ;)

rfield19, 2021-09-19:
Awesome, thanks - I'll have a dig around...  Have a great evening!

sn4k3, 2021-09-21:
Fixed

closed 2021-09-21 by sn4k3