sn4k3/UVtools #308: [BUG]  Light off time is different, when ctb format is converted to pwmx(photon format).

sjhan12, 2021-09-29:
## System
 - OS: [e.g. Windows 10 x64 v20H2]
 - Slicer:  chitu 1.8.1
 - Printer: 

## Describe the bug

After I generate .ctb file with bottom delay 5s and normal delay 5s in chitubox 1.8.1, 

".ctb file" is converted to .pwmx file through UVtools2.23.1. 

However, it shows Light-off (12.05s) with UVtools vesion2.23.1.  

It doesn't show 5s in Light-off for Photon monoX.  

I check the .pwmx file with photon workshop. it shows 12.05s in photonworkshop. 

I attached the files link https://www.dropbox.com/sh/4ouugke4ndarewe/AADEIXXGSx2Qu0kTXVBfwlT9a?dl=0  , you can find the files. 

Please check this bug. 

Thanks

## To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-09-29:
Might be your Automation settings, check that:

[image]

sjhan12, 2021-09-29:
Thank you for comments. I will try the method which you told. I will get back with results. 

I guess the Light-off delay doesn't work properly. 

When I choose "light off to zero", I got light-off 0 second in pwmx. but with other options, I got 12.05s. 

https://www.dropbox.com/sh/4ouugke4ndarewe/AADEIXXGSx2Qu0kTXVBfwlT9a?dl=0 

I attached the files with "option name". Please check this.

sn4k3, 2021-09-29:
> When I choose "light off to zero", I got light-off 0 second in pwmx. but with other options, I got 12.05s.

That is correct. You must understand what light-off delay is in order to understand this, all other options will calculate the right value based on settings + extra time: https://github.com/sn4k3/UVtools/wiki/Rest-times-and-TSMC

sjhan12, 2021-09-29:
Thank you for fast comments. 

I will go over the document and get back to you. Thanks again.

sn4k3, 2021-11-13:
Fixed

closed 2021-11-13 by sn4k3