sn4k3/UVtools #309: [FEATURE REQUEST] Integration into Prusa Slicer

uwwint, 2021-10-04:
Hi there, 

I have some leave upcoming and am tinkering with the idea of integrating the UVTools conversion functionality into Prusa Slicer. Prusa Slicer issue is here: https://github.com/prusa3d/PrusaSlicer/issues/5280 . I have some leave upcoming to work on this. Are you interested in tackling this together? 

KR,
Uwe

sn4k3, 2021-10-04:
At this point it not worth to have just a simple convertion after export.
It's much more beneficial to open the SL1 file on UVtools, let it auto convert and search for issues and verification.
Additional if you wish to just convert without the UI, you can create a desktop shortcut for UVtools with command: --convert And then drag and drop SL1 files into that.

But if you want to enhancement PrusaSlicer that is ok. Having the option to run command per printer after file slice/export is a thing i dont understand why dont exists already, but then they have it per print profile on FDM... 

Regarding UVtools and file formats if you have any question i can fill that gap, but i have no wishes to Fork PS and add that extra burden in my life at this point.

sn4k3, 2021-10-10:
Current implementation is not usable, as i told, this should exist also per printer.
Per resin profile is a pain to configure as profiles can be shared...

mtaumike, 2021-10-17:
> Current implementation is not usable, as i told, this should exist also per printer.
> Per resin profile is a pain to configure as profiles can be shared...

Hmm.. wonder if adding a more 'global' post-processing scripting system could be handy, then just checkbox the materials you want run through post-processing.

closed 2021-10-29 by sn4k3