sn4k3/UVtools #310: [BUG]

eddy-99, 2021-10-06:
## System
 - OS: Windows 10 Version	10.0.19043 Build 19043
 - Slicer: Chitubox
 - Printer: Orange 10

## Describe the bug
Pixel editor tab hidden

## To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior
When the program is open in full screen pixel editor tab disappears. Resizing the window revels the tab.
Great program. Thanks.
## Screenshots

## Files

 
## Exceptions Log file

## Additional context

tslater2006, 2021-10-06:
Unable to reproduce locally, launched UVTools and maximized the window. Pixel Editor tab is visible

[image]

sn4k3, 2021-10-06:
This is due screen resolution scaling. Other than native 100% resolution is a plague.
See more here: https://github.com/sn4k3/UVtools/issues/112

PS: You pixel editor is located behind "Rotate" you still be able to click it if you click right below the rotate
Bug is UI framework related and scaling