sn4k3/UVtools #333: [BUG] Recursively remove islands non-functional

hodgeac, 2021-10-29:
## System
 - OS: Windows 10 x64 v21H1
 - Slicer: Lychee Slicer
 - Printer: Voxelab Proxima

## Describe the bug
When attempting to repair layers and issues, option to "Recursively remove island for up to:" does not seem to function at all regardless of value used.

## To Reproduce
Steps to reproduce the behavior:
1. Go to 'Repair Layers and issues'
2. Click on 'field for "Recursively remove islands for up to:" and set to anything. Default is 4, zero is supposed to remove all islands recursively. None of those options is functional.'
3. Make sure your islands are smaller than the defined "Remove islands equal or smaller than:" value.
4. Click 'Repair layers and issues button'
5. Only a single layer island will be repaired with no recursive behavior.
6. Repeat over and over and over for every single layer with an island.

## Expected behavior
As the tooltip for the "Recursively remove islands for up to:" option defines, remove islands on each layer above if the removal of an island on the current layer results in an island on the layer(s) above up to the define number of layers.

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-10-29:
Works for me, what's your UVtools version?

hodgeac, 2021-10-29:
2.23.6 X64. Maybe it's down to the file type? My printer uses .fdg files. I know it's fairly uncommon.

hodgeac, 2021-10-29:
General configuration for OpenCV 4.5.3 =====================================
  Version control:               4.5.3-234-g78f1119f43

  Extra modules:
    Location (extra):            D:/bb/cv_x64/build/build_x64/../opencv_contrib/modules
    Version control (extra):     4.5.3-60-gd0ad2609

  Platform:
    Timestamp:                   2021-07-27T16:30:01Z
    Host:                        Windows 10.0.19043 AMD64
    CMake:                       3.20.0
    CMake generator:             Visual Studio 16 2019
    CMake build tool:            C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
    MSVC:                        1929
    Configuration:               Debug Release MinSizeRel RelWithDebInfo

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x64/cl.exe  (ver 19.29.30038.1)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MD /O2 /Ob2 /DNDEBUG 
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MDd /Zi /Ob0 /Od /RTC1 
    C Compiler:                  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30037/bin/Hostx64/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP   /MD /O2 /Ob2 /DNDEBUG 
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP /MDd /Zi /Ob0 /Od /RTC1 
    Linker flags (Release):      /machine:x64  /INCREMENTAL:NO 
    Linker flags (Debug):        /machine:x64  /debug /INCREMENTAL 
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          wsock32 comctl32 gdi32 ole32 setupapi ws2_32 D:/bb/cv_x64/build/build_x64/install/lib/freetype.lib D:/bb/cv_x64/build/build_x64/install/lib/harfbuzz.lib D:/bb/cv_x64/build/build_x64/install/lib/libhdf5.lib vtkRenderingOpenGL2 vtkCommonCore vtksys vtkCommonDataModel vtkCommonMath vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkRenderingCore vtkCommonColor vtkCommonComputationalGeometry vtkFiltersCore vtkFiltersGeneral vtkFiltersGeometry vtkFiltersSources vtkglew vtkInteractionStyle vtkFiltersExtraction vtkFiltersStatistics vtkImagingFourier vtkImagingCore vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOCore vtkdoubleconversion vtklz4 vtklzma vtkzlib vtkFiltersTexture vtkRenderingFreeType vtkfreetype vtkIOExport vtkIOImage vtkDICOMParser vtkmetaio vtkjpeg vtkpng vtktiff vtkIOXML vtkIOXMLParser vtkexpat vtkRenderingContext2D vtkRenderingGL2PSOpenGL2 vtkgl2ps vtkIOGeometry vtkIOLegacy
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib quirc

  OpenCV modules:
    To be built:                 alphamat aruco barcode bgsegm bioinspired calib3d ccalib core datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hdf hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking ts video videoio videostab viz wechat_qrcode xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    python_bindings_generator python_tests world
    Disabled by dependency:      -
    Unavailable:                 cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev cvv java julia matlab ovis python2 python3 sfm
    Applications:                perf_tests
    Documentation:               NO
    Non-free algorithms:         NO

  Windows RT support:            NO

  GUI: 
    Win32 UI:                    YES
    VTK support:                 YES (ver 8.2.0)

  Media I/O: 
    ZLib:                        build (ver 1.2.11)
    JPEG:                        build-libjpeg-turbo (ver 2.1.0-62)
    WEBP:                        build (ver encoder: 0x020f)
    PNG:                         build (ver 1.6.37)
    TIFF:                        build (ver 42 - 4.2.0)
    JPEG 2000:                   build (ver 2.4.0)
    OpenEXR:                     build (ver 2.3.0)
    HDR:                         YES
    SUNRASTER:                   YES
    PXM:                         YES
    PFM:                         YES

  Video I/O:
    DC1394:                      NO
    FFMPEG:                      YES (prebuilt binaries)
      avcodec:                   YES (58.134.100)
      avformat:                  YES (58.76.100)
      avutil:                    YES (56.70.100)
[... 11 lines cut ...]
  Other third-party libraries:
    Eigen:                       YES (ver 3.3.7)
[... cut ...]
  Python (for build):            C:/python-virt/python37/python.exe

  Java:                          
    ant:                         NO
    JNI:                         C:/Program Files/Java/jdk1.8.0_202/include C:/Program Files/Java/jdk1.8.0_202/include/win32 C:/Program Files/Java/jdk1.8.0_202/include
    Java wrappers:               NO
    Java tests:                  NO

  Install to:                    D:/bb/cv_x64/build/build_x64/install
-----------------------------------------------------------------

Is that valuable info?

hodgeac, 2021-10-29:
BTW I don't even have a D: drive so that D:/bb/... path is invalid.

sn4k3, 2021-10-29:
Can you send a file that i can reproduce the problem?

hodgeac, 2021-10-29:
[triangleusingcyl (2).zip](https://github.com/sn4k3/UVtools/files/7444661/triangleusingcyl.2.zip)

sn4k3, 2021-10-29:
Ok i saw the problem with that single pixel, i will see what is happening here

sn4k3, 2021-10-30:
Found the problem, will be fixed on next patch

hodgeac, 2021-10-30:
You're awesome. Thanks!

sn4k3, 2021-11-03:
Fixed

closed 2021-11-03 by sn4k3