sn4k3/UVtools #337: [BUG] Menu bar options don't stay open on MacOS

bgyarfas, 2021-11-04:
## System
 - OS: Mac OS 11.6
 - Slicer: NA
 - Printer: NA
 - UVtools version: 2.24.1

## Describe the bug
When clicking any of the menu items (File, Tools, Calibration, etc) they will flash for a moment and then disappear. I confirmed this is only with the 2.24.1 release.

sn4k3, 2021-11-04:
Looks like a bug on current UI framework which have been updated to the new version on this release, i will report this to them.

bgyarfas, 2021-11-04:
Done! Thanks!

tirendel, 2021-11-04:
I just confirmed this with version 2.24.0 as well. Running MacOS Monterey (Version 12.0.1); MacBook Pro 15-inch, 2017

sn4k3, 2021-11-04:
> I just confirmed this with version 2.24.0 as well. Running MacOS Monterey (Version 12.0.1); MacBook Pro 15-inch, 2017

2.24.0 is almost the same as 2.24.1 so bug is present on both. Downgrade while we wait for a fix from Avalonia

sn4k3, 2021-11-14:
This was fixed by UI framework version downgrade to 10.8

closed 2021-11-14 by sn4k3