sn4k3/UVtools #338: Incorrect Image Ratio 2.24 Latest

eddy-99, 2021-11-07:
## System
 - OS:Windows 10
 - Slicer: Prusa
 - Printer: Elegoo mars 2
 - UVtools 2.24

## Describe the bug
Just updated from 2.23 to latest 2.24 and this error message is displayed when I import a SL1 file from Prusa Slicer. Reverted back to 2.23 and all is good.
[image]

## To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-11-07:
Why are you ignoring a UVtools warning and **crutial information** and thinking it's a bug?
Read the message carefully, this is a life saver!

You have 1620px in X but a size of 127mm while you have 2560px in Y but 80mm
Just correlate resolution with display size. So, you have larger resolution with smallest size. In other words your lcd screen size are swap! So yeah, your ratio is wrong and will lead to malformed model.
You should  use my profiles via Profile manager on UVtools instead, your mars 2 config should look like:

[image]

eddy-99, 2021-11-07:
Cool. 2.23 Just worked  so  I thought problems.
Thanks Eddy

[prusa-slicer_2021-11-07_15-38-49]<https://user-images.githubusercontent.com/113281/140651710-b263bf99-3a10-4f84-8cea-3914251c7905.png>

closed 2021-11-07 by sn4k3