sn4k3/UVtools #343: [BUG] Layer slider is super laggy if you drag it

Lenbok, 2021-11-13:
## System
 - OS: Ubuntu 20.04
 - Slicer: PrusaSlicer
 - Printer: Elegoo Mars 2
 - UVtools version: 2.24.2

## Describe the bug

If you grab the layer slider grip (blue ball thingy on the right) and drag it up and down, it's very laggy (probably it is trying to show every single layer even if by the time the layer is shown the slider is well into another area?). This is most visible if you grab the slider and drag it fast up and down a few times, then let go of the slider, the UI replays the whole movement over the next few seconds. The workaround is to never drag the slider, instead click to where you want to go, but that slider grip is big and tempting if that's where your habits lie.

## To Reproduce
Steps to reproduce the behavior:
1. Load a decent sized file, analyze it
2. Click and hold on the layer slider grip (blue ball thing)
3. Drag the layer slider fairly fast up and down from top to bottom a few times and let go of the slider
4. Wait while the UI replays all those layers

## Expected behavior
When dragging the layer slider, once one layer has finished displaying, the next layer to be shown should correspond to where the layer slider/thumb is *right then*.

tslater2006, 2021-11-13:
I think your "expected" is actually describing the current behavior. For
every increment that's changed its trying g to render the layer as the
slider is *right then* even if mid scroll. I would say the expected
behavior would be to not render a layer until the drag has finished. This
would prevent the intermediate layers from trying to show.

Lenbok, 2021-11-14:
Yes, sorry, I meant the expected next layer when dragging was where the mouse (rather than slider thumb) is right then. Your suggestion of not rendering the layer until the drag has finished is probably the easiest way to remedy the problem although you would want the layer indicator to still update, so that while you are dragging you can still see the height/layer/percent that you are dragging to.

sn4k3, 2021-11-14:
I know that on linux the slider behaves a little diferent, at lest on my VM i got the same slowness. 
On windows it jumps if you attempt to drag fast. Each image render time can been seen on bottom right with the "ms" label.
If each image takes much to load then it's normal to freeze the scroll.

A reset timer can fix this but will give the impression of breaks on fast render machines with windows.
I can take a look

sn4k3, 2021-11-23:
Implemented, Check on Settings -> Layer preview -> The last input box

closed 2021-11-23 by sn4k3