sn4k3/UVtools #344: Error opening the file

JoaoPaul54, 2021-11-14:
When trying to open up a photons file I get this error.
I tried to re-slice it and even updated my chitubox with no change.

The file is a big one with a bit over 50Mb, not sure if that matters.

System.AggregateException: One or more errors occurred. (Error image ran shortly or off the end, expecting 3686400 pixels, got 3686273 pixels.)
 ---> System.IO.FileLoadException: Error image ran shortly or off the end, expecting 3686400 pixels, got 3686273 pixels.
   at UVtools.Core.FileFormats.PhotonSFile.LayerDef.Decode(Boolean consumeRle) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonSFile.cs:line 233
   at UVtools.Core.FileFormats.PhotonSFile.<>c__DisplayClass98_0.<DecodeInternally>b__0(Int32 layerIndex) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonSFile.cs:line 531
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.TaskReplicator.Run[TState](ReplicatableUserAction`1 action, ParallelOptions options, Boolean stopOnFirstFailure)
   at System.Threading.Tasks.Parallel.ForWorker[TLocal](Int32 fromInclusive, Int32 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.ThrowSingleCancellationExceptionOrOtherException(ICollection exceptions, CancellationToken cancelToken, Exception otherException)
   at System.Threading.Tasks.Parallel.ForWorker[TLocal](Int32 fromInclusive, Int32 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](TSource[] array, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body)
   at UVtools.Core.FileFormats.PhotonSFile.DecodeInternally(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonSFile.cs:line 504
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 2586
   at UVtools.WPF.MainWindow.<>c__DisplayClass102_0.<ProcessFile>b__1() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1377

[image]

sn4k3, 2021-11-14:
Some pixels are missing on layer image... Can you send me your file?

JoaoPaul54, 2021-11-14:
I can but i'd rather not post it here since it's a paid stl and this is a public repository.
I'll send it to your email with a wetransfer link ok?

btw, just noticed you are Portuguese like me :P

sn4k3, 2021-11-14:
That will work to me.
Yes i'm portuguese.

sn4k3, 2021-11-14:
I found the problem, it is the same problem i had on other anycubic file formats. 
Sometimes slicer may not fill all remaning black pixels, ending shortly, this is kinda random, but can be seen as an space optimization, but the randomess make no sense...

Will fix on next patch

sn4k3, 2021-11-14:
Fixed

closed 2021-11-14 by sn4k3

JoaoPaul54, 2021-11-14:
Many thanks for the file back. Also good to know it was a bug and not something stupid from my side xD
Great job on the app,