sn4k3/UVtools #347: [BUG]

serenata-kenneth, 2021-11-18:
Trying to open the photon file here https://github.com/altLab/photon-resin-calibration/blob/master/resin-xp-finder/resin-test-50u.B100.2-20.photon but I keep getting the error showing in the screenshot 

[image]

sn4k3, 2021-11-18:
It's not a bug, it's a safe-fail check error.
Error is clear... Layer 2 is positioned at 4mm Z  while layer 3 is positioned at 0.1mm Z. So layer 3 have lower Z than layer 2... 
File is malformed and have bad layer data, then UVtools warn  you and prevent the file from open.

All layers with exception of layer 2 are positioned at 0.1mm, so that layer is causing trouble.

[image]

serenata-kenneth, 2021-11-19:
Understood. 

I wanted to load up the file in UVtools so that I can adjust the resolution since i recently modded my OG Photon by adding a new LCD screen from ChituSystems with a different resolution.

sn4k3, 2021-11-19:
The only way is to bypass that security check.
Tell me the resolution and i will change the file for you, but only if LCD screen have same dimension as the original, otherwise test is no longer valid and require a new slice.

serenata-kenneth, 2021-11-20:
The lcd is exactly the same size since it is an upgrade 5.5 inch replacement 

The new resolution is 1920 x 1080

Many thanks!

sn4k3, 2021-11-20:
Humm thats looks like an downgrade...
Original file is 1440x2560
LCD: 67.5 x 120 mm
Pixel: 46.87um

With your new LCD you will have a pixel size of: 62.5um and then the slice image will be no longer valid, even if i adjust the resolution you will cure pixels 1.33x bigger than should yield in a larger bloated test, can only solve this by reslice the file with correct machine info. OR get other file that ends in a pixel um similar to 62.5um, target machine doesnt matter as soon you got the res and display right.
But if you don't mind in streching the test i can do the resolution modification for you

serenata-kenneth, 2021-11-20:
Well yes a downgrade in terms of resolution, but an upgrade in terms of exposure times and lcd lifespan because from 8 seconds exposure I went down to around 1.4 seconds  :)

I wouldn't mind in stretching the test. Appreciate the help!

sn4k3, 2021-11-20:
Ok, i was unable to change resolution and fit all content:

[image]

So i resized images by 66.66% to fit the content and match your pixel um. 
[resin-test-62.5u.B100.2-20_FULLHD.zip](https://github.com/sn4k3/UVtools/files/7575052/resin-test-62.5u.B100.2-20_FULLHD.zip)

Hope that fits your needs

serenata-kenneth, 2021-11-20:
Excellent! Thank you

closed 2021-11-20 by sn4k3