sn4k3/UVtools #351: [BUG] Halot lite model doesn't match printer

dmEngineering, 2021-11-29:
## System
 - OS: [e.g. Windows 10 x64 v20H2]
 - Slicer: Halot Box
 - Printer: Halot Lite CL-89L
 - UVtools version: 2.52.2

## Describe the bug
After opening a .cxdlp file sliced with Halot Box, I set up the accuracy test and save it in the USB drive.
I import the file into the Halot Lite (firmware: 2.201.20) with no problem, then when I try to print it gives me the following message: "This Model does not match the printer!"

Lychee should release an update with halot lite in the next weeks, maybe this will help you.
Let me know if it is better to wait for the Lychee update.

## Files

[Tollerance_print in place.zip](https://github.com/sn4k3/UVtools/files/7618990/Tollerance_print.in.place.zip)

sn4k3, 2021-11-29:
I will add support for that

sn4k3, 2021-12-02:
Implemented

closed 2021-12-02 by sn4k3

rebeltaz, 2022-12-17:
I am having the same issue. Is there something I need to change in the settings or something? Or do I need to update the firmware on the printer? I've tried using version 2 and 3 of the file format but both give the same "does not match" results using v3.9.2 of UVtools.

mikecarr, 2023-03-26:
I am seeing the same issue but with Halot Lite when using Prusa Slicer. I noticed there is a difference in printer model when using the SL1 files vs HalotBox, SL1 PrinterModel is CL-89, where as HalotBox is CL-89L. I am not sure if this is something I can change and where. 

UVtools: 3.12.1
PrusaSlicer: 2.5.1

rebeltaz, 2023-03-26:
This is what snk3 told me and it worked, so... give it a shot? ¯\_(ツ)_/¯

```
Ok I found why it didn't work. They have two machines with same resolution and UVtools get the first which is the CL-89, but yours it's CL-89L raising that problem.
To fix that after converting you can go to File -> Terminal, and send this two commands:

SlicerFile.MachineName = "CL-89L";
SlicerFile.Save();

```

mikecarr, 2023-03-26:
A few more out of sync values:

PrinterModelSize = is 6 should be 7
PrinterModelArray = is 6 should be 7

[image]

sn4k3, 2023-03-26:
@mikecarr they are on sync. PrinterModel is null terminated which means after CL-89L there is a 0x00 byte which adds +1 in size

mikecarr, 2023-03-26:
Argh ok, I was still getting the same error. After I run the previous commands do I need to do anything else? Or will those changes save to the file?

mikecarr, 2023-03-26:
Debug:2023-03-26 22:08:00:ModelFileData LoadModelFile: begin parse cxdlp file
Debug:2023-03-26 22:08:00:subModelFileData.s_FileLayerCount is  48
Debug:2023-03-26 22:08:00:gDlpPrintData->s_X is  "192"
Debug:2023-03-26 22:08:00:gDlpPrintData->s_Y is  "120"
Debug:2023-03-26 22:08:00:gDlpPrintData->s_Thickness is  "0.05"
Debug:2023-03-26 22:08:00:gDlpPrintData->MaxLayer size is  48
Debug:2023-03-26 22:08:00:gDlpPrintData->s_Versioncode is  3
Debug:2023-03-26 22:08:00:strMachine is  "CL89"  g_machineInfo.name is  CL89L
Debug:2023-03-26 22:08:00:strMachine is not queal g_machineInfo.name "CL89" CL89L
Debug:2023-03-26 22:08:00:SerialPortPrintFile::run: failed to parse file.

sn4k3, 2023-03-26:
> After I run the previous commands do I need to do anything else? Or will those changes save to the file?

You need to run that every new file you open with UVtools. After run those two commands that file will always be correctly saved by UVtools, unless you convert it.

mikecarr, 2023-03-27:
Ok that is working, I was sending the file to the wrong folder. I would be nice if there was an option to bookmark or set macros to send and save commands like this :-) Excellent work on this tool

mikecarr, 2023-03-27:
> > After I run the previous commands do I need to do anything else? Or will those changes save to the file?
> 
> You need to run that every new file you open with UVtools. After run those two commands that file will always be correctly saved by UVtools, unless you convert it.
> 
> You still have the problem where name does not match: `Debug:2023-03-26 22:08:00:strMachine is "CL89" g_machineInfo.name is CL89L`
> 
> As alternative make PrusaSlicer machine name only as `CL-89L` or use my machine preset from UVtools

Yes I did use your machine preset.