sn4k3/UVtools #352: [FEATURE REQUEST] Add "Open With External Application" to save menu.

ccatlett1984, 2021-12-02:
My example would be Lychee slicer, which after slicing offers me the option to open a file in UVTools for more processing, or another external application. (I specify the command line arguments).

It would be nice to be able to "chain" the file from UVTools into other applications, without having to save/close UVTools and manually open the file in the external app.

sn4k3, 2021-12-02:
I can add that option, allowing to select a application to open the file with

sn4k3, 2021-12-02:
Proposal:

[image]

ccatlett1984, 2021-12-02:
Amazing, thank you.

Can we get a setting option to specify the name, path to file, and
arguments to pass?

sn4k3, 2021-12-02:
You will be able to pick the application to execute and provide arguments to pass. On arguments you can pass {0} to indicate the loaded file.

So you can do crazy stuff like: cmd.exe "{0}" "C:\mynewfile.whatever"

ccatlett1984, 2021-12-02:
Awesome 👍

sn4k3, 2021-12-02:
Configuration

[image]

sn4k3, 2021-12-02:
Implemented

closed 2021-12-02 by sn4k3