sn4k3/UVtools #353: [BUG] "Pattern loaded model" with "multi-exposure: Embedded text does not reflect different exposure times.

ccatlett1984, 2021-12-02:
When using the "Pattern loaded model" option in UVTools, and the multiple exposure option, the "text" that is embedded into the bottom of the models does not reflect the different exposure times.
[image]
[image]

sn4k3, 2021-12-02:
That numbers are actually right.
You are using multiple brightness, meaning all the layer will be exposed for 120s/12.75s but some section will expose will more grey pixels than the others, the time it get exposed is still 120s/12.72s regarding the grey level.
The time you see on multiple brightness table, is just a simulation of the time if the convertion was 1:1 (brightness to time) which is not in most case, only printers with fake AA can use this times as reference and still, it need to be multiples of what printer AA supports.

You should never follow that time numbers on a real AA printer. The only number that matters here is the grey level (0 to 255) which shows correctly

ccatlett1984, 2021-12-02:
I understand that they are just "estimates" of what the timings might be.
I wish my printer supported multiple exposures (ctb v2) but it doesnt.
I guess i can reference the estimate by the grey level, and get my target time, then start printing single calibration parts to dial it in.

sn4k3, 2021-12-02:
What is your printer?

> I guess i can reference the estimate by the grey level, and get my target time, then start printing single calibration parts to dial it in.

That will be very hard and require either a sensor and/or many tests.
This test as you have it only tell you if you found a good grey level, you can use it with that exposure time, meaning you have to grey all the model to that level and print with the base exposure time (120s/12.75s) to get the same results from the test. But grey do not correlate with given time on the table

ccatlett1984, 2021-12-02:
Oh, i understand now.
Thanks for taking the time to explain that so thoroughly.

closed 2021-12-02 by ccatlett1984