sn4k3/UVtools #354: [BUG]

bgyarfas, 2021-12-02:
## System
 - OS: macOS 11.6
 - Slicer: Lychee 3.6.0
 - Printer: Elegoo Satturn
 - UVtools version: 2.52.2

```
1. Go to "UVtools -> Help -> About"
2. Click on "Copy information"
3. Paste that infomation bellow this line:
UVtools v2.25.2
Operative system: Darwin 20.6.0 Darwin Kernel Version 20.6.0: Mon Aug 30 06:12:21 PDT 2021; root:xnu-7195.141.6~3/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz
Processor cores: 16
Memory RAM: 15.90 / 31.47 GB
Runtime: osx.11.0-x64
Framework: .NET 5.0.12
AvaloniaUI: 0.10.8
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 200% (Primary)
    WA: 2560 x 1415    UA: 1280 x 708
2: 1280 x 800 @ 200%
    WA: 1280 x 775    UA: 640 x 388
3: 1440 x 2560 @ 200%
    WA: 1440 x 2535    UA: 720 x 1268

Path: /Applications/UVtools.app/Contents/MacOS

General configuration for OpenCV 4.5.4 =====================================
  Version control:               4.5.4-235-g27567928d7

  Extra modules:
    Location (extra):            /Users/tiago/Desktop/emgucv/opencv_contrib/modules
    Version control (extra):     4.5.4-61-gc33e5b60

  Platform:
    Timestamp:                   2021-11-01T19:19:02Z
    Host:                        Darwin 18.7.0 x86_64
    CMake:                       3.21.4
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3 SSSE3 SSE4_1
      requested:                 DETECT
    Dispatched code generation:  SSE4_2 FP16 AVX AVX2 AVX512_SKX
      requested:                 SSE4_1 SSE4_2 AVX FP16 AVX2 AVX512_SKX
      SSE4_2 (1 files):          + POPCNT SSE4_2
      FP16 (0 files):            + POPCNT SSE4_2 FP16 AVX
      AVX (4 files):             + POPCNT SSE4_2 AVX
      AVX2 (30 files):           + POPCNT SSE4_2 FP16 FMA3 AVX AVX2
      AVX512_SKX (5 files):      + POPCNT SSE4_2 FP16 FMA3 AVX AVX2 AVX_512F AVX512_COMMON AVX512_SKX

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                /Library/Developer/CommandLineTools/usr/bin/c++  (ver 10.0.1.10010046)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /Library/Developer/CommandLineTools/usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,-dead_strip  
    Linker flags (Debug):        -Wl,-dead_strip  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          -framework OpenCL -framework Cocoa -framework AppKit /Users/tiago/Desktop/emgucv/platforms/macos/build/install/lib/libhdf5.a Iconv::Iconv
[... 168 lines cut ...]
## Describe the bug
Lychee 3.6.0 now allows you to export different versions of CTB (v2-v4). UVTools errors when trying to open a v4 version of a file with the following error:

```
System.IO.FileLoadException: Malformed file, PrintParametersV4 found invalid validation values, expected (4, 4) but got (0, 0)
File name: '/Users/brett/elegoo_saturn/To Print/iC Dev Fixture v1.v4.ctb'
   at UVtools.Core.FileFormats.ChituboxFile.DecodeInternally(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\ChituboxFile.cs:line 2043
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 2625
   at UVtools.WPF.MainWindow.<>c__DisplayClass104_0.<ProcessFile>b__1() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1490
```

## To Reproduce
Steps to reproduce the behavior:
1. Change the version of the CTB file being export by lychee by modifying your printer settings in lychee (Under "Other Parameters")
2. Export a v4 file
3. Open in UVTools
4. See error

## Expected behavior
Since UVtools has support for v4 I thought it'd be able to open it. I did try to rename the file to `.v4.ctb` but it still reported the same error.

## Additional context
UVTools is still able to open a CTB v3 file as exported by Lychee

Let me know if you need anymore information! Thanks for all the great work!

sn4k3, 2021-12-02:
Spotted on Atlas discord too, already addressed.
Will be good on next release

sn4k3, 2021-12-02:
Fixed

closed 2021-12-02 by sn4k3