sn4k3/UVtools #356: [BUG] UVTools does not start under fesh instllation of Windows 11

MeierAlex, 2021-12-03:
UVTool does not start under Windows 11. i also tried st start it from the commandline -> No output.

## System
 - OS: Windows 11 N + Media Extension Pack
 - Slicer: 
 - Printer: 
 - UVtools version: v2.25.3

```
1. Go to "UVtools -> Help -> About"
2. Click on "Copy information"
3. Paste that infomation bellow this line:

```

## Describe the bug
A clear and concise description of what the bug is.

## To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-12-03:
The problem is the Windows **N** variant. Even with Media Extension Pack seens not enough to make OpenCV run.
If you are on a fresh install i recommend you install  the normal version, the non-N and it will fix your problem

MeierAlex, 2021-12-03:
Thank you very much for your reply - I sadly do not have the opportunity to use an other version than Windos 11N. Do you have a hint for me how I cn get UVTools to work ?

sn4k3, 2021-12-03:
> Thank you very much for your reply - I sadly do not have the opportunity to use an other version than Windos 11N. Do you have a hint for me how I cn get UVTools to work ?
Would it be possible to simply install OpenCV ?

I have no clue... Others with Windows 10 N + Media Extension Pack also failed to get it to work. It's not UVtools related. 
Your best shot is asking for help on correct repository: https://github.com/emgucv/emgucv/discussions and link with this issue
Something else is failing other than the Media Extension Pack requirement

MeierAlex, 2021-12-03:
I have started a discussion - thank you very much. it would be a pitty if I could not use you brilliant software any more :/

sn4k3, 2021-12-03:
> I have started a discussion - thank you very much. it would be a pitty if I could not use you brilliant software any more :/

It will be solved, is a matter of time and discover what is missing...

MeierAlex, 2021-12-03:
Fixed the problem - you have to install the media extionsions via windows by "Add optional feature" -> "Media Extensions"

sn4k3, 2021-12-03:
Thanks for discover that, i will update readme accordingly

closed 2021-12-03 by sn4k3