sn4k3/UVtools #361: [BUG] Issue with CTBv4 from Lychee with TSMC set off (possibly Lychee bug)

hafriedlander, 2021-12-04:
## System

 - OS: Windows 10 x64 v20H2
 - Slicer: Lychee 3,6,0
 - Printer:  Elegoo Saturn
 - UVtools version: 2.25.3

## Description

Lychee 3.6.0 when slicing CTBv4 has two options:

```
Two Stage Motion Control (TSMC) On/Off
Light On Delay (On) / Resting Delay (Off)
```

When TSMC is on, everything works fine.

When TSMC is off, there are two bugs

1) The Bottom Retract Speed summary displays as 0

This appears to be because Lychee is setting the PrintParametersV4->BottomRetractSpeed to 0 and setting the retract speed per layer (like the old CTBv3 behaviour)

```
Print Parameters V4 -> BottomRetractSpeed: 0, BottomRetractSpeed2: 0, Padding1: 0, Four1: 0, Padding2: 0, Four2: 0, RestTimeAfterRetract: 0, RestTimeAfterLift: 0, RestTimeBeforeLift: 0, BottomRetractHeight2: 0, Unknown1: 1684.6448, Unknown2: 9740, Unknown3: 0, LastLayerIndex: 303, Padding3: 0, Padding4: 0, Padding5: 0, Padding6: 0, DisclaimerAddress: 712, DisclaimerLength: 320, Reserved: System.Byte[]
-Image GROUP 0-
```

2) The RestTime values are always displayed as 0 (even when Resting Delay is set to On, and values provided)

This appears to be a bug in Lychee, the values output in debug during loading show 0

```
LAYER 18 -> LayerDef: PositionZ: 0.95, ExposureTime: 4.1, LightOffSeconds: 7.83, DataAddress: 216269, DataSize: 1427, Unknown1: 0, TableSize: 84, Unknown3: 0, Unknown4: 0, TotalSize: 1511, LiftHeight: 5, LiftSpeed: 40, LiftHeight2: 0, LiftSpeed2: 0, RetractSpeed: 40, RetractHeight2: 0, RetractSpeed2: 0, RestTimeBeforeLift: 0, RestTimeAfterLift: 0, RestTimeAfterRetract: 0, LightPWM: 100
```

## To Reproduce

Export any file in Lychee 3.6.0 with CTBv4 and TSMC _OFF_

## Expected behavior

BottomRetractSpeed and RestTime are displayed correctly

## Sliced files

Correct (TSMC on)

[tsmc_on.zip](https://github.com/sn4k3/UVtools/files/7653255/tsmc_on.zip)

Incorrect (TSMC off)

[tsmc_off.zip](https://github.com/sn4k3/UVtools/files/7653256/tsmc_off.zip)

```
UVtools v2.25.3
Operative system: Microsoft Windows 10.0.19042 X64
Processor: Intel(R) Core(TM) i9-10980XE CPU @ 3.00GHz
Processor cores: 36
Memory RAM: 41.85 / 63.68 GB
Runtime: win10-x64
Framework: .NET 5.0.10
AvaloniaUI: 0.10.8
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

[... 184 lines cut ...]
081: System.Diagnostics.Tracing, Version=5.0.0.0
082: System.Diagnostics.DiagnosticSource, Version=5.0.0.0
083: System.Net.Security, Version=5.0.0.0
084: System.Security.Cryptography.X509Certificates, Version=5.0.0.0
085: BinarySerializer, Version=8.6.1.2
086: System.Threading.Tasks, Version=5.0.0.0
087: System.Text.Encoding, Version=5.0.0.0
088: System.IO, Version=5.0.0.0
089: System.Reflection.TypeExtensions, Version=5.0.0.0
090: System.Reflection, Version=5.0.0.0
091: System.Reflection.Extensions, Version=5.0.0.0
092: System.Net.Sockets, Version=5.0.0.0
093: System.Net.NameResolution, Version=5.0.0.0
094: System.IO.Compression.ZipFile, Version=5.0.0.0
095: System.Collections.Specialized, Version=5.0.0.0
096: System.Security.Cryptography.Primitives, Version=5.0.0.0
097: System.Security.Principal.Windows, Version=6.0.0.0
098: System.Security.Claims, Version=5.0.0.0
099: System.Security.Principal, Version=5.0.0.0
100: System.Threading.Tasks.Parallel, Version=5.0.0.0
101: System.Security.Cryptography.Encoding, Version=5.0.0.0
102: System.Runtime.Numerics, Version=5.0.0.0
103: System.IO.FileSystem.DriveInfo, Version=5.0.0.0
104: Microsoft.Win32.Registry, Version=6.0.0.0
```

sn4k3, 2021-12-04:
Yes, not UVtools related.
All fields are coming from Lychee and untouched by UVtools.
Please forward the problem to Lychee team

hafriedlander, 2021-12-06:
I've flagged with Lychee, but didn't see a response from them. No reason to leave this open though. Closing for now.

closed 2021-12-06 by hafriedlander