sn4k3/UVtools #364: [BUG]

LigaMoscow, 2021-12-05:
## System
 - OS: [e.g. Windows 10 x64 v20H2]
 - Slicer: Lichee slicer 3(3.5.1)
 - Printer: Anycubic Photon mono x
 - UVtools version: 2.25.3

```
1. Go to "UVtools -> Help -> About"
2. Click on "Copy information"
3. Paste that infomation bellow this line:
UVtools v2.25.3
Operative system: Microsoft Windows 10.0.19042 X64
Processor: Intel(R) Core(TM) i7 CPU         930  @ 2.80GHz
Processor cores: 8
Memory RAM: 5.03 / 5.99 GB
Runtime: win10-x64
Framework: .NET 5.0.12
AvaloniaUI: 0.10.8
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1857 x 1080    UA: 1857 x 1080

Path: C:\Program Files\UVtools

General configuration for OpenCV 4.5.4 =====================================
  Version control:               4.5.4-235-g27567928d7

  Extra modules:
    Location (extra):            D:/bb/cv_x64/build/build_x64/../opencv_contrib/modules
    Version control (extra):     4.5.4-61-gc33e5b60

  Platform:
    Timestamp:                   2021-10-27T09:08:07Z
    Host:                        Windows 10.0.22000 AMD64
    CMake:                       3.21.3
    CMake generator:             Visual Studio 16 2019
    CMake build tool:            C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
    MSVC:                        1929
    Configuration:               Debug Release MinSizeRel RelWithDebInfo

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe  (ver 19.29.30136.0)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MD /O2 /Ob2 /DNDEBUG 
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MDd /Zi /Ob0 /Od /RTC1 
    C Compiler:                  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP   /MD /O2 /Ob2 /DNDEBUG 
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP /MDd /Zi /Ob0 /Od /RTC1 
    Linker flags (Release):      /machine:x64  /INCREMENTAL:NO 
    Linker flags (Debug):        /machine:x64  /debug /INCREMENTAL 
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          wsock32 comctl32 gdi32 ole32 setupapi ws2_32 D:/bb/cv_x64/build/build_x64/install/lib/freetype.lib D:/bb/cv_x64/build/build_x64/install/lib/harfbuzz.lib D:/bb/cv_x64/build/build_x64/install/lib/libhdf5.lib vtkRenderingOpenGL2 vtkCommonCore vtksys vtkCommonDataModel vtkCommonMath vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkRenderingCore vtkCommonColor vtkCommonComputationalGeometry vtkFiltersCore vtkFiltersGeneral vtkFiltersGeometry vtkFiltersSources vtkglew vtkInteractionStyle vtkFiltersExtraction vtkFiltersStatistics vtkImagingFourier vtkImagingCore vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOCore vtkdoubleconversion vtklz4 vtklzma vtkzlib vtkFiltersTexture vtkRenderingFreeType vtkfreetype vtkIOExport vtkIOImage vtkDICOMParser vtkmetaio vtkjpeg vtkpng vtktiff vtkIOXML vtkIOXMLParser vtkexpat vtkRenderingContext2D vtkRenderingGL2PSOpenGL2 vtkgl2ps vtkIOGeometry vtkIOLegacy
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib quirc

  OpenCV modules:
    To be built:                 alphamat aruco barcode bgsegm bioinspired calib3d ccalib core datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hdf hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking ts video videoio videostab viz wechat_qrcode xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    python_bindings_generator python_tests world
    Disabled by dependency:      -
    Unavailable:                 cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev cvv java julia matlab ovis python2 python3 sfm
    Applications:                perf_tests
    Documentation:               NO
    Non-free algorithms:         NO
[... 165 lines cut ...]

```

## Describe the bug

all layers with the "dynamic lift" option have "
LiftHeight NaNmm @ NaNmm / min "
**The error occurred only in this project.**

## To Reproduce
Steps to reproduce the behavior:

## Expected behavior

## Screenshots

https://drive.google.com/file/d/1ymV-v2uZAOFWopiIRuaTJXlpzhdJk4p1/view?usp=sharing
## Files

https://drive.google.com/file/d/1TVXRNDdIujqCn1e-kpcZLLkbQDFM3zj-/view?usp=sharing

 
## Exceptions Log file

## Additional context

sn4k3, 2021-12-05:
Even with your values i can't replicate the problem.
When i run the tool Lifts get back to normal and correct value...
Can you provide the configuration? 

Go to tool, configure to reproduce this, click on menu button on bottom left, export settings and attach here

sn4k3, 2022-02-10:
Since i was not able to replicate this and no more feedback, i'm closing. Reopen on new evidence.

closed 2022-02-10 by sn4k3