sn4k3/UVtools #368: Phrozen Sonic Mini 4K - memory error

JasonBelec, 2021-12-11:
App opens, everything works, export to USB for testing, goto to printer, start and Memory Error. Every time, tried playing with settings, no go.

sn4k3, 2021-12-11:
Upload your file here for i to check

JasonBelec, 2021-12-11:
[PhrozenSonicMini4K_1.8_2.6_0.2.ctb.zip](https://github.com/sn4k3/UVtools/files/7697548/PhrozenSonicMini4K_1.8_2.6_0.2.ctb.zip)

Had to ZIP it...

sn4k3, 2021-12-11:
The machine name is wrong, from where do you created that .ctb?
If chitubox you need to select the right machine from the list and not a default created by you

JasonBelec, 2021-12-11:
Good catch. I loaded the default test file from the manufacturer (Phrozen) which would use Chitubox. I use Mango.

sn4k3, 2021-12-11:
That's a bad pratice. You should always load a sliced file by you, one that you print previous by preference.
Retry with my suggestion and report back.

JasonBelec, 2021-12-11:
Ok, All good, thanks for your speedy support. I must have missed that did bit of information, my bad.

closed 2021-12-11 by JasonBelec